hzeller/rpi-rgb-led-matrix #175: Issues running the examples

codingjoe, 2016-07-15:
Hi @hzeller ,

thanks for providing this package! Since I mostly code in high level languages like Python, I'm not very familiar with cool hardware hacking and appreciate every help I can get.

So I bought a Adafruit HAT for the RPi. When I clone their repo und run the example, it works great!
When I want to run your examples, it doesn't work. Which I really don't get since their examples use your driver.

As you might already expect, I'm interested in the Python bindings. Is there a secret ingredient I'm missing to make it work?

Btw: I have a 64x32 matrix (2 chained 32).

Cheers,
Joe

hzeller, 2016-07-15:
The Readme has a section about the HAT; you need to set an option while compiling.

codingjoe, 2016-07-16:
@hzeller was that added recently or am I just blind as a bat 🙈  😆 
Anyways, thanks, I'll give it a try on Monday!

closed 2016-07-16 by codingjoe