hzeller/rpi-rgb-led-matrix #176: Printing and scrolling text to whole height

DZDomi, 2016-07-15:
I am trying to print a text to the whole screen of my 32x32 board and also make it scrollable, like the example:

`led-matrix -D 1 runtext.ppm`

The example is working fine for me, but the problem is that I am not able to create my own custom scrolling text. The python api only prints a smaller size version of the text (~half of the screen). I also tried to create, via various netbpm commandos, my own scrolling .ppm files but i had the same problem as above. How was the runtext.ppm produced? Are there any ways this is possible?

RcArtSolutions, 2016-08-01:
You can use Photoshop or any other software to generate a bitmap-file (e.g. *.png) with a heigh of 32px. Use the online converter (https://convertio.co/image-converter/) to convert it into *.ppm.

DZDomi, 2016-08-01:
Thanks for you answer, the problem is, that i need the generate the .ppm images on the fly. Basically convert the output from one programm to a .ppm scrolling text and then send it to the led matrix.

hzeller, 2016-08-02:
You can just write a little program that uses the text-api to scroll text.

DZDomi, 2016-08-02:
Thank you for your awesome library and for the reply.

In the Text-Api i need to specify a font, but the problem is, that the fonts do not fit to the whole height of the screen, there is always padding on bottom and top.

hzeller, 2016-08-02:
Couldn't you generate a font of the right height and store it as BDF file ?

DZDomi, 2016-08-02:
If you could point me in the right direction, for generating such fonts, are there any fonts available to download? How where the fonts in this repo created?

RcArtSolutions, 2016-08-03:
I am sorry, I misunderstood your question.

To generate BDF files you can use [FontForge](https://github.com/fontforge/fontforge), a easy to use font editor. This editor allows also the automatic generation of bitmap-based fonts, but the results are not very good. So I render the glyphs in Photoshop and import it to FontForge.

I modify the [bdf-font.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/bdf-font.cc) for perfect display fonts with various glyph-width (monospace).

Expand the glyph-struct with "d_width":
`struct Font::Glyph {
  int width;
  int height;
  int y_offset;
  int d_width;  // width incl. spacing behind glyph.
  rowbitmap_t bitmap[0];  // contains 'height' elements.
};`

Read in the "d_width" (drawing width) in the "LoadFont" method (after parsing the encoding):
`else if (sscanf(buffer, "DWIDTH %d %d", &tmp.d_width, &dummy) == 2) {
  // parsed.
}`

And change the return value of the "DrawGlyph" method to the d_width:
`return g->d_width;`

I am not sure where is the issue, it is also possible that my generated BDF files are incorrect. But for me work it, with the d_width-change.

In my case I display only numeric values, so I have only BDF files with the glyphs 0..9.
[image]
An example with the BDF version of a free font called "Bebas Neue".

DZDomi, 2016-08-03:
Wow! This helps enormously! Thank you very much! 

It basically worked, except that some of the letters are not showing. I used your trick and changed the code in the bdf-font.cc and also converted different ttf and otf fonts. I tried it with FontForge and also the freeware command line tool [otf2bdf](https://www.math.nmsu.edu/~mleisher/Software/otf2bdf/) but always get the same result. Some of the letters are not showing. I made a video to show you what I mean:

https://www.youtube.com/watch?v=wgGtxvEIzN0

The font "Bebas Neue" is the only one, that, after trying out different settings, was able to show all letters, but the font does not support lowercase letters. :/ Any suggestions on fixing this problem?

RcArtSolutions, 2016-08-03:
I know this issue, too. Check the _BBX_ x-offset for the invisible glyphs.

I have another modification in the "bdf-font.cc" to kill the x-offset automatically. I use the _FOUNDRY_ element to identify a FontForge BDF file. I think it is also possible to detect files there was generated with oft2bdf.

See the "bdf-font.cc" in my forked repo: https://github.com/RcArtSolutions/rpi-rgb-led-matrix/blob/master/lib/bdf-font.cc

hzeller, 2016-08-04:
@RcArtSolutions can you send me a pull request with the more fixed bdf-font.cc ? I think I only ever used it for fixed fonts, and my implementation was very sloppy. So any improvements are welcome :)

RcArtSolutions, 2016-08-04:
@hzeller I want to find out where is the issue. In my generated (FontForge) BDF files or in the _LoadFont_ method. I will test it for my self and read some docs about the BDF format. After this, I am going to send a pull request, no problem.

I work also at a bitmap version of "Bebas Neue" with alphanumeric charset (only capital letters).

frostiebot, 2017-03-16:
Don't mean to bump this issue, but after using the changes suggested by @RcArtSolutions I went from oddball glyph rendering to essentially perfect rendering of bdf fonts generated from a ttf source through FontForge.

I wanted to also add that killing the X Offset if the FOUNDRY value is "FontForge" is not necessarily always the right thing to do. I was converting one "monotype" truetype font and for almost all cases FontForge generated the right values for BBOX, but in the case of glyphs that were "centered" inside the bounding box, DWIDTH had to be modified to correct the glyph positioning.

Not entirely sure where I'm going with this, but I wanted to give a +1 to the changes by @RcArtSolutions, as at the very least it fixed a _lot_ of weird issues I was encountering with bdf fonts generated by FontForge off the bat.

guigobass, 2017-05-14:
Hello...

my english is not good...sorry.

i dont understand this problem....

i program in python and i need up size font. i have a panel 64x32 and i can't up de font size.

how i can do this? help me please

hzeller, 2017-05-14:
You need to get a font of the size you need @guigobass . The fonts used here are fixed size fonts that can't be scaled (scaled fonts are not a good idea in very low resolution environments like these).

guigobass, 2017-05-15:
i understand you, i don,t know very much this, is first week that i work this....i need to do one query to SQL server, this is ok but my problem is the size....exist some thing who i can do this...general ppm for pass text dinamic? i need thath for each query pass one message with number, this number change you understand me? so, as my size panel is 64 x 32 y need that my text  ocupe more y size....you know who do? you have any example, exist some thing...please help me.

guigobass, 2017-05-15:
a need a size shuch as raspberry pi example ./demo -D1 runtext.ppm but dinamic....

hzeller, 2017-05-15:
Please make an effort to read the API

closed 2017-07-30 by hzeller