hzeller/rpi-rgb-led-matrix #178: Scramble Matrix

shoke2k, 2016-07-18:
Hello, 
When I run the test -D 1 
`sudo ./led-matrix -r 16 -D 1 runtext16.ppm`
I get this:
[(https://youtu.be/5rmsz7zKhHo)]
Do you know what is the problem?

depili, 2016-07-18:
Might be a 1/4 scan panel or something like in this issue: https://github.com/hzeller/rpi-rgb-led-matrix/issues/136

shoke2k, 2016-07-18:
I think that is the scan, but I try with options -r 8 and -c 2 and not working (at readme said that hasn't be tested )

hzeller, 2016-07-18:
Yes, that looks like it is essentially working with -r8 -c2.
If you look at the pattern, you see that you just need to write a [transformer](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/transformer.h)
top map the pixel layout of your panel. Two lines rows are made to form four rows by 'reflecting' it at the end of the panel. So your transformer needs to undo that.

shoke2k, 2016-07-21:
The Scan rate for this board is 1/2 according the web of the manufacturer.

hzeller, 2016-08-10:
do you have a link to that Matrix ? Without any information it is hard to help.

shoke2k, 2016-08-10:
https://www.alibaba.com/product-detail/High-quality-import-grade-outdoor-3535_60493048771.html?spm=a2700.7724857.0.0.3bVfIY
This is the module.
I try with anothers modules of others manufacturers, 32x32, 16x32 and 16x16, and I've only gotten that works well with the 16x16 module.

rodrigosalinas, 2016-08-23:
Hi @shoke2k. I have almost the same problem than you. Instead of having lines of 2 rows back and forth like you, I have lines of 4 rows.

Based on the issue #136 pointed out above, I wrote my own Transformer. First I had to figured out the correlation between the expected coordinates and the real ones in my panel. For that I used the test-pixel published in the issue #136 by @KentWalker. I wrote it down one by one in a spreadsheet with 4 columns: X, Y expected and real X, Y. All the 512 coordinates. Like this:

x	y		x'	y'
0	0		7	0
1	0		6	0
2	0		5	0
3	0		4	0
4	0		3	0
5	0		2	0
6	0		1	0
7	0		0	0
8	0		23	0
9	0		22	0
10	0		21	0
11	0		20	0

And so on.

From those values, I had to discover the pattern and to write it in an algorithm to transform the first X, Y values into the last ones. I got to this (sorry by the name, not met to offend anybody):

    /******************************************/
    /* Crazy Chinese Panel Transformer Canvas */
    /*           by Rodrigo Salinas           */
    /******************************************/
    class CrazyChineseTransformer::TransformCanvas : public Canvas {
    public:
        TransformCanvas() : delegatee_(NULL) {}

        void SetDelegatee(Canvas* delegatee);

        virtual void Clear();
        virtual void Fill(uint8_t red, uint8_t green, uint8_t blue);
        virtual int width() const;
        virtual int height() const;
        virtual void SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue);

    private:
        Canvas *delegatee_;
    };

    void CrazyChineseTransformer::TransformCanvas::SetDelegatee(Canvas* delegatee) {
        delegatee_ = delegatee;
    }

    void CrazyChineseTransformer::TransformCanvas::Clear() {
        delegatee_->Clear();
    }

    void CrazyChineseTransformer::TransformCanvas::Fill(uint8_t red, uint8_t green, uint8_t blue) {
        delegatee_->Fill(red, green, blue);
    }

    int CrazyChineseTransformer::TransformCanvas::width() const {
        return delegatee_->width() / 2;
    }

    int CrazyChineseTransformer::TransformCanvas::height() const {
        return delegatee_->height() * 2;
    }

    void CrazyChineseTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
        // HERE IT GOES YOUR OWN ALGORITHM TO TRANSFORM THE COORDINATES 
        // GENERATED BY THE LIBRARY INTO THOSE EXPECTED BY THE PANEL
        int new_x = 0;
        int new_y = 0;
        if (y < 4 || (y > 7 && y < 12)) {
            new_x = (x * -1) + (24 * ((x / 8) + 1)) - 17;
            new_y = y;
        } else {
            new_x = x + (8 * ((x / 8) + 1));
            if (x < 24) {
                new_y = y - 4;
            } else {
                new_y = (y == 4 || y == 12) ? y : y - 4;
            }
        }
        delegatee_->SetPixel(new_x, new_y, red, green, blue);
    }

    /***********************************/
    /* Crazy Chinese Panel Transformer */
    /*        by Rodrigo Salinas       */
    /***********************************/
    CrazyChineseTransformer::CrazyChineseTransformer() 
        : canvas_(new TransformCanvas()) {}

    CrazyChineseTransformer::~CrazyChineseTransformer() {
        delete canvas_;
    }

    Canvas *CrazyChineseTransformer::Transform(Canvas *output) {
        assert(output != NULL);

        canvas_->SetDelegatee(output);
        return canvas_;
    }

The previous lines of code you have to add them to the ```lib/transformer.cc``` file. And of course you also have to update the ```include/transformer.h``` file adding these lines:

    class CrazyChineseTransformer : public CanvasTransformer {
    public:
        CrazyChineseTransformer();
        virtual ~CrazyChineseTransformer();

        virtual Canvas *Transform(Canvas *output);

    private:
        class TransformCanvas;

        TransformCanvas *const canvas_;
    };

After cleaning and compiling all the library again, I can execute the demo and obtain the expected but with one problem. It works fine with one panel, but with two I can't scroll the image from one panel to the other. Instead I have the images duplicated in both panels.

I hope what I have until now can help somebody with the same problem than I. But at the same time I hope somebody can help me to achieve to make it work in more than one panel.

Thanks in advance.

snktnaik, 2017-03-15:
Hi,
i have understood the transformer.h and transformer.cc part of the process
but do i need to edit the main code file to make the transformation work

closed 2018-02-25 by hzeller