hzeller/rpi-rgb-led-matrix #179: LED panel availability on aliexpress.com

rhildinger, 2016-07-18:
Does anyone know what happened to all the RGB LED panel vendors on aliexpress.com? All of the sudden (within the past 30 days) there are very few panels of any size available, and most of the vendors that I have ordered from in the past seem to have completely stopped selling anything. Anyone have any ideas what might have happened?

depili, 2016-07-20:
Seems like all sellers that used the adafruit images for their product have indeed disappeared. Probably due to the images used. They will be back when the sellers manage to get new pictures, probably.

rhildinger, 2016-07-20:
I contacted four of the sellers that I have bought from int he past, and they all claim that it is due to some sort of "catalog reclassification" or aliexpress reformation. However, I think you are right depili - I think aliexpress is trying to polish up their image a bit and they are cracking down on vendors misrepresenting their products or using copyrighted pictures from other vendors. I suspect Adafruit may have threatened legal prosecution over the use of their images. I think every single LED panel vendor on that site was using Adafruit's pics.

jpoppe, 2016-07-25:
Keyes robot, Robotdyn (Nice high quality Arduino suppliers) and most of the OLED stores also disappeared. Robotdyn is at least back. I am also waiting for the led panel stores to come back. It seems they are still on Alibaba.

I bought my first panel @ Leeman Group, and got a panel with a dead (loose) pixel and getting practically no support from them and the dispute time already passed before I tested the panel.

Would be awesome if people could post recommended stores here. Since Adafruit panels ridiculously expensive in Europe.

hzeller, 2016-08-10:
If found that search-terms such sa P4 or P5 and RGB or LED and HUB75 typically yields some useful results.

jpoppe, 2016-08-11:
It seems most stores are back, Leeman told me to send a replacement display with my next order, after a couple of messages, so service seems pretty OK in the end. I have ordered displays from 3 different suppliers. Will share my experiences here as soon as I get them up and running. Time to start soldering my controller board ;).

BTW thanks a lot for your nice library and controller layouts!

closed 2016-08-23 by hzeller