hzeller/rpi-rgb-led-matrix #180: Increasing number of panels in chain leads to brightness reduction

marcinzdunek, 2016-07-19:
Hello,
Currently I am trying to use your library to display text using 5 32x32 LED panels. Unfortunately I don't have any information about scanning of these panels. I tried different modes and this is what I get:
- rows = 8, subpanels = 2, chain = 2 - brightness is good, but I can light up only 1-4, 9-12... rows, no way to remap pixels using transformer or anything else. Just rows 5-8, 13-16 and so on don't want to light up.
- rows = 16, subpanels = 2, chain = 2 - brightness is lower, I can build up proper letters using transformer:
```
int new_x, new_y;
new_x = ((x / 16) * 32) + (x % 16);
if ((y & 8) == 0)
    new_x += 16;
new_y = ((y / 16) * 16) + (y % 8);
delegatee_->SetPixel(new_x, new_y, red, green, blue);
```
With rows = 16 brightness using 1 panels is not bad, but when chaining 2-5 panels it is getting worse and worse with each panel. Using 5 panels and PWM bits = 1 I get 250 MHz on my Raspberry Pi 2. I am using white (255,255,255) color to test it.
Rows = 32 gives the same result, only brightness is even lower than with rows = 16.
Additionally I got driver and software which is provided by Chinese manufacturer. Using this driver and software I can get 100% brightness even with 5 panels. So I guess that it's possible on Raspberry Pi 2 too...
With 1 panel I can get about 96% of brightness from Chinese driver.
With 2 panels about 90%
With 3 panels about 80%
With 4 panels about 70% and so on...
Increasing number of panels in chain leads to brightness reduction.

What can I do to achieve the same brightness? What am I doing wrong? Or maybe this is issue with library?

hzeller, 2016-07-19:
It sounds like you indeed have a 16 row/2 sub-panel matrix. Driving it with 16 rows is necessarily darker than with 8, because it is multiplexing. 

Clocking data into panels is slower than the lowest PWM time the library generates, so there is necessary black time. All these things are necessary a compromise and the library settings are conservative defaults that worked with a wide range of them.

It will of course be different than dedicated hardware that is supposed to work with a particular kind of panel.

Some points to play with:

 * The primary reason for longer chains to be darker is, that clocking in the data is taking more time. Comment out the `DEFINES+=-DRGB_SLOWDOWN_GPIO=1` line in `lib/Makefile` (add a `#` in front) to create a higher clock-rate, but not all RGB panels are actually fast enough to deal with it, so you have to try that with yours.
 * You can increase the kBaseTimeNanos in [lib/framebuffer.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer.cc#L38). This will reduce refresh rate but increase the relative on-time.
 * If you have increased kBaseTimeNanos, you can up the refresh rate again by using less PWM bits. The 11 bits this library is doing is probably already beyond what the driver board you got is doing. Reducing it to 10 probably is not too bad.
 * You really should run the panels in parallel; there is [an easy-to-build PCB](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) already in the project. Running boards in parallel uses more GPIO pins, but it uses the same amount of CPU. But now you have shorter chains.
 * Hopefully, you are not using the Adafruit HAT, but if you do, you need to do the hardware modification described in the README to enable overlapping clocking and output enable which is an important part of getting more brightness.
 * Make sure to also test with other image content than just a white image when compare the two solutions. It is very simple to cheat in cases with uniform-all-on light to not clock in new data to avoid dark time. (Which but that is not what this library is doing, mostly because there is no point in real life and you want a non-fluctuating refresh rate; but we don't know about that from the driver solution you have).

hzeller, 2016-07-19:
It is not entirely clear from your test-setup, but don't use PWM=1 for the all-white test because that is the worst case scenario: the on-time is 130ns, while the clock-time is several microseconds, so you only get a fraction of the light. If you are using PWM=1, then it is expected to have a large hit in brightness for multiple panels.

So use higher PWM numbers or increase the kBaseNanos.

(There is also one known hardware limitation in the PWM generation that can result into about 1/8 less brigthness than the best possible case, but I don't think that that is what you are seeing).

marcinzdunek, 2016-07-20:
Hello hzeller,
I have tried running application one more time using your advices.
I set PWM to 10 and kBaseNanos to different values between 100 and 300.
I can't run panels in parallel, because in the nearest future I need to run 6x3 panels.
I set slowdown to different options.
Unfortunately panels are still much less bright than using Chinese driver. 
I need to use your library, because Raspberry Pi is highly needed in my project.
What else can I do? Is there a way to get 6 panels as bright as if I connect one?

depili, 2016-07-20:
Couldn't the library just clock the next data in, then disable output, latch the data and enable output, thus preventing the black time on data input?

marcinzdunek, 2016-07-20:
Here are oscilloscope photos:

Chinese driver CLK
[image]

Raspberry CLK
[image]

Chinese driver RGB pins
[image]

Raspberry RGB pins
[image]

All photos have been generated using 5 panels, r=16, subpanels=2, chain=10
Why there are so many differences in duty cycles?

hzeller, 2016-07-20:
@depili the clocking and output-enable are already interleaved, but it won't buy much in the lower bits as the on-time is much shorter (< 200ns) as the clock time (number of horizontal pixels * 50ns). It pays off for higher bit-planes.

@marcinzdunek thanks for the measurements, that is very nice. The clock of the Chinese driver looks suspicious: why is it only in the order of 100us ? This would only work if it is really only black/white with a slow update. Does it do full 33Bit color like the Pi library ?

marcinzdunek, 2016-07-20:
Yes, it does show full 33-bit palette of colors with high brightness on even 6 panels in chain. I am currently trying to find any differences between this Chinese driver and yours library. Unfortunately I don't have an access to code, this software from Chinese driver allows me only to input text, image or promitives and run it. But what about RGB duty cycle? Isn't that solution of so low brightness?

hzeller, 2016-07-20:
No, if the Chinese driver clock is in the order of 10kHz (100usec), then it does _not_ do 33 bit color in parallel. At best, it uses one color out of a palette.
It only can do that low clock-rate if it shows one brightness color at a time.

In order to compare, you have to put gray scale wedge on both configurations, something like
[image]
Otherwise you are not comparing the same thing.

The brightness is proportional to the duty cycle, correct.
With longer panels, you have to clock more data. For the lower bits, that means that the clocking takes longer than the overlapping on time. Say with 20Mhz clock time, you need 1.6usec per panel to clock in data, but the on-time is only 130ns in the lowest bit. If you have 5 panels, you spend 8usec clocking in data, but only switch on for 130ns. Just after the 6th bit, the overlapping on-time is longer than clocking in the next cycle.

You can't have both with these kind of pixels: high color resolution and full brigthness (unless you want a refresh-rate of 2Hz or so :) ).

marcinzdunek, 2016-07-20:
Okay, so what is the way to achieve full brightness with only 8 color palette? 8 colors is enough for me, brightness matters the most

hzeller, 2016-07-20:
Can you try setting the PWM bits (`matrix.SetPWMBits(1)`) to one and measure with the oscilloscope the signal on the OutputEnable pin ?

marcinzdunek, 2016-07-20:
Right now I don't have an access to oscilloscope, but when I have connected OE to one LED it was blinking rapidly

marcinzdunek, 2016-07-21:
So there is not any way to improve multiple panels' brightness using your library?

hzeller, 2016-07-21:
there probably is. Did you do the measurement?

hzeller, 2016-07-22:
I have now added an experimental option in `lib/Makefile`, that you can enable by uncommenting `#DEFINES+=-DEXPERIMENTAL_HIGH_BRIGHTNESS` and recompile.

I have not tried this yet as i am away from my panel set-up currently, so please let me know what you see.

hzeller, 2016-07-23:
With recent changes, even the normal mode should be fairly good. I've also more documented the nanosecond parameter and its trade-offs; also it became a define in the Makefile (`LSB_PWM_NANOSECONDS`).
For very long chains (like 30 or so), the `-DEXPERIMENTAL_HIGH_BRIGHTNESS` mode will still give a slight advantage in brightness, but then you only really can use PWM=1.

marcinzdunek, 2016-07-25:
Unfortunately nothing has changed, I am testing for 5 panels with different parameters:
DRGB_SLOWDOWN_GPIO from 0 to 3
DLSB_PWM_NANOSECONDS from 100 to 300
DEXPERIMENTAL_HIGH_BRIGHTNESS on and off.
And result is still the same.

hzeller, 2016-07-25:
You should essentially get almost no noticable loss in brightness from 1 to 10 panels. In particular if you have a low PWM value. The output-enable is essentially always on except when switching rows.
If you don't see a difference - did you make sure to re-compile everything, not only the library, but also your program linked fresh against the library ?

marcinzdunek, 2016-07-25:
Yes, I have just downloaded your fresh library, write only displaying each line instead of text in text-viewer.cc, changed options in lib/Makefile and run. No effects.

hzeller, 2016-07-25:
Do you have an output of your of what `make` outputs when compiling and paste it here ?

You essentially now should have the maximum the displays can do also there should be no fall-off in brightness between 1 or 10 displays.

If you measure with PWM=1, you'll see that output-enable is essentially triggered all the time. If your displays are very slow (the Chinese driver indicates that) that they can only do lower frequencies to turn full on, increase the nanoseconds to 1000 or even 2000.

marcinzdunek, 2016-07-25:
I have downloaded this file from Raspberry through PSCP to Windows, so I am not sure if its coding is good. Anyway I am sending that executable, just change extension from PDF to none.

[prog.pdf](https://github.com/hzeller/rpi-rgb-led-matrix/files/381009/prog.pdf)

marcinzdunek, 2016-07-25:
Setting DLSB_PWM_NANOSECONDS to 1000 makes the same brightness and framerate drops to 60 MHz

marcinzdunek, 2016-07-25:
[image]

hzeller, 2016-07-25:
re prog.pdf: I don't mean the resulting binary, the output of what make writes to the terminal.

I am going to bed, so will be offline for a few hours. You should measure the output of our output-enable signal, it should essentially be enabled except for very brief peaks now. At least your display should now be as bright as with the Chinese driver. But it can't make it brighter :)

marcinzdunek, 2016-07-25:
Ah so there is make's result:

[image]

hzeller, 2016-07-25:
Alright, compile-output looks good. With that, things should work fine and no noticable fall-off in brightness between `-c1` and `-c6`.

Make sure your power-supply is beefy enough (for each 6 panels you need about 20Amps) and that the voltage, measured at the panels when they are full white is 5V (you need pretty thick and short cabeling for that). If your cables have too much inductance make sure to add capacitors as it won't be able to supply the current spikes.

marcinzdunek, 2016-07-25:
Cables connecting Chinese and Raspberry drivers are the same. Is it possible, that lower voltage on Pi's pins is the reason of this lower brightness?

marcinzdunek, 2016-07-25:
I have discovered that commenting setting transformer line gives perfect brightness which is not dependent of chain parameter. Why using transformer causes lower brightness?

marcinzdunek, 2016-07-25:
New version of library helped with brightness correction. But I have an issue now with transformer so I am starting another issue and closing this one.

closed 2016-07-25 by marcinzdunek