hzeller/rpi-rgb-led-matrix #183: Using custom transformer leads to brightness reduction on multiple panels

marcinzdunek, 2016-07-25:
While using this transformer:

```
int LargeSquare64x64Transformer::TransformCanvas::width() const {
  return delegatee_->width() / 2;
}

int LargeSquare64x64Transformer::TransformCanvas::height() const {
  return delegatee_->height();
}

void LargeSquare64x64Transformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  if (x < 0 || x >= width() || y < 0 || y >= height()) return;
  int new_x, new_y;
  new_x = ((x / 16) * 32) + (x % 16);
  if ((y & 8) == 0) {
    new_x += 16;
  }
  new_y = ((y / 16) * 16) + (y % 8);
  delegatee_->SetPixel(new_x, new_y, red, green, blue);
}
```

with 5 panels, rows = 32, chain = 10, subpanels = 2 and color white I am trying to display 160x32 filled matrix. But brightness is much higher when I use the same parameters and disable this transformer.
Of course this transformer allows me to display all the letters properly, but first I want to display filled white 5 panels as bright as possible. 
Why is enabling this transformer leading to less bright display?

hzeller, 2016-07-25:
So if you have 5 32x32 panels (which results in 160x32 pixel), why do you have to to use `-c10 -r32` ? That would account for 320x32 pixels, but then you would need to connect 10 panels.

Are you using `SwapOnVSync()` by any chance ? Be aware that the off-screen canvases don't automatically go through the transformer (issue #102) unless you explicitly say so:
>transformer()->Transform(the_swap_canvas)->SetPixel()`.

Depending on the transformer, you might only unintentionally fill part of the pixels in that case, and with rapid switching between the buffers, it looks dimmer, because only one is white, the other has only part of the pixels set.

marcinzdunek, 2016-07-25:
With c = 5 only 2,5 of panel is filled, but brightness is still worse than without setting transformer.
I am enabling transformer here in text-example.cc:
```
RGBMatrix *canvas = new RGBMatrix(&io, rows, chain, parallel);
  canvas->SetTransformer(new LargeSquare64x64Transformer());
```
What else can I do?

hzeller, 2016-07-25:
And then you use `canvas->Fill(255, 255, 255);` to get all white I presume ?

You need to better understand the mapping of your panels. If you have `-c10 -r32`, you are describing 320x32 pixel.

marcinzdunek, 2016-07-25:
Instead of DrawText line I use:
```
for(int i=0;i<32;i++)
    DrawLine(canvas, 0, i, 159, i, color);
```
and then with -c5 I got only 2 and half of panel filled.

marcinzdunek, 2016-07-26:
When I use -r 16 and sub panels = 2 I am able to display correct brightness using transformer, but only top half of panels can be light up. With sub panels = 1 brightness is lower (as low as in 1st post).

marcinzdunek, 2016-07-26:
My transformer was written badly and I need to use -c16 option, now everything is great, thanks for the help!

hzeller, 2016-07-26:
Cool!

closed 2016-07-26 by hzeller