hzeller/rpi-rgb-led-matrix #184: Refresh rate

bagumondigi, 2016-07-26:
Hi @hzeller , how much is refresh rate in your library ? Is it configurable ? 
Thanks you.

bagumondigi, 2016-07-26:
i've uncomment DEFINES+=-DSHOW_REFRESH_RATE and it return 166.2Hz. Is it dependence on how many LED panel i used ? If i want to slowdown the refresh rate to a constant number, how can i do that ?

marcinzdunek, 2016-07-26:
Hello, I need to do exactly the same thing. I need to obtain constant refresh rate

hzeller, 2016-07-26:
First off: you don't want to get much lower than that, because you start to see flicker. Human eyes are really sensitive to flicker in particular for this high contrasts. Ideally, you'd like to have as much refresh rate as possible.

Why do you want to _reduce_ the refresh rate ?

The refresh rate depends on various factors (number of LED panels, multiplexing, slowdown-gpio, PWM-nanoseconds, PWM-Bits). You can't really set the final rate, but you can modify these values to tweak the rate to your liking. Since you can't do much about the first parameters, you can tweak the PWM-nanoseconds (higher numbers: lower refresh rate). Lower PWM bits increase the refresh rate a bit, but low PWM numbers only really make sense for lower number of colors.

The refresh rate will not really be a constant, because it depends on how busy the operating system is, but it typically hovers within a narrow range (make sure to use the `isolcpus` configuration as described in the README to have these things scheduled on a free CPU).

marcinzdunek, 2016-07-26:
I got functions which are trying to move static text (and some additional operations within it). While displaying static text on 5 panels I got framerate about 470 Hz. But when I move this text my framerate dropss to different values 300-400. And then this scrolling text slows down. I need to have my text scrolling at constant speed.

bagumondigi, 2016-07-26:
@marcinzdunek Did you try runtext.py in python/samples of this library ? It include a simple scrolling text effect that work well and write in python so easy to edit it.
@hzeller thanks for great answer, I just think that we only need 24 frames per seconds to view smoothly.

hzeller, 2016-07-26:
This looks like a hefty drop for the framerate. I would only expect that with hardware timing generation disabled, like with `DISABLE_HARDWARE_PULSES` or `EXPERIMENTAL_HIGH_BRIGHTNESS` ( @marcinzdunek  you might still use the last one from your previous experiments, maybe you want to disable it, as it doesn't buy much brightness with your six panels, but it will heavily influence the variability of your framerate).

@bagumondigi 24 Frames is painful flickering, you really _don't_ want that. You want _at least_ 100Hz for somewhat pleasing viewing experience. Better 150Hz and up.

There is a difference in 'animation framerate' and 'display refresh rate'. For animations, 24 pictures/s are enough. But for display, you need much higher refresh rate (in film-cinemas with 24 fps film, they make it possible by  showing the same image two to four times, so that the display flicker rate is 48 or 96 hz).

So if you display stuff, just generate a new frame every few V-syncs.

(which is a good idea: I should add a framerate-fraction as a parameter to SwapOnVSync() to make that simple).

bagumondigi, 2016-07-26:
How can we check how many CPU and RAM was used ?

hzeller, 2016-07-26:
Just use the usual Unix utilities for that, @bagumondigi (ps, top).

marcinzdunek, 2016-07-26:
Thank you very much, I will try manipulating some constants.

closed 2016-07-27 by bagumondigi