hzeller/rpi-rgb-led-matrix #185: led video viewer

AmagiZuiho, 2016-07-29:
Hello, I'm trying to show the video to the LED screen but do not know where to start. Does anyone have done on this issue yet?

closed 2016-08-04 by AmagiZuiho

hzeller, 2016-08-04:
You could do something based on http://dranger.com/ffmpeg/tutorial01.html

Or, what I do these days, is to compile FlaschenTaschen with the [RGB matrix backend](https://github.com/hzeller/flaschen-taschen/tree/master/server#rgb-matrix-panel-display), and then use either

a) a [VNC bridge]( https://github.com/scottyallen/flaschenvnc) to stream it from any machine on the network.

b) the current [development VLC](https://wiki.videolan.org/UnixCompile/) contains a [FlaschenTaschen backend](https://git.videolan.org/?p=vlc.git;a=commit;h=cf334f257868d20b6a6ce024994e84ba3e3448c3) - with that, you then can play video with something like this:
vlc --vout flaschen --flaschen-display=hostnameofdisplay \
      --flaschen-width=160 --flaschen-height=96 \
      <video-filename-or-YouTube-URL>

hzeller, 2017-05-28:
I have now included a video viewer that natively can play on the LED matrix:

https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#video-viewer