hzeller/rpi-rgb-led-matrix #188: Run the runText outside of sample folder not working

serverfire, 2016-08-06:
First of all , thank you for great work .

When i'm running the runtext.py everything work , but when write this code , i get error say i need pass the canvas , it is same code but outside the sample folder 

```
from .core import RGBMatrix, FrameCanvas
import graphics
import time

def RunText():
    matrix = RGBMatrix(32, 6, 1)
    matrix.pwmBits = 11
    matrix.brightness = 100

    offscreenCanvas = matrix.CreateFrameCanvas()

    font = graphics.Font()

    font.LoadFont("../matrix/fonts/7x13.bdf")

    textColor = graphics.Color(255, 255, 0)

    pos = offscreenCanvas.width
    myText = "Hello World!"

    while True:
        offscreenCanvas.Clear()
        len = graphics.DrawText(offscreenCanvas, font, pos, 10, textColor, myText)
        pos -= 1
        if (pos + len < 0):
            pos = offscreenCanvas.width
        time.sleep(0.05)
        offscreenCanvas = matrix.SwapOnVSync(offscreenCanvas)
```

i get this error : 

```
Argument 'c' has incorrect type (expected rgbmatrix.core.Canvas, got rgbmatrix.core.FrameCanvas)

```

hzeller, 2016-08-06:
Paging @Saij as I don't know how the Python bridge works.

Saij, 2016-08-06:
Well normaly FrameCanvas should inherit Canvas. Have to take a look into it.

hzeller, 2016-08-10:
Could it be something like shared object files not properly resolved ?

serverfire, 2016-08-13:
other example of problem , 

When go to sample folder and do this : 
```
python runtext.py
```
It work , but when you go to parent folder 
```
cd ..
python sample/runtext.py 

```

won't work , same problem as my code

serverfire, 2016-08-13:
only runtext not working other code that are using only the core , they work

hzeller, 2016-08-13:
Did you check if the font is properly loaded ? Maybe the Python impl changes the working directory, so maybe an absolute path to the font might be something to try.

serverfire, 2016-08-14:
@hzeller problem of not working from upper folder fixed with font path , thank you

closed 2016-08-23 by hzeller