hzeller/rpi-rgb-led-matrix #191: OSH Park board -- is that a full assembly or just a board ready for soldering?

yarikoptic, 2016-08-11:
Sorry for a naive question: the board from OSH Park, is that just the board itself, so I would need to purchase/solder all components  or it comes "fully assembled" as on e.g. https://github.com/hzeller/rpi-rgb-led-matrix/raw/master/img/three-parallel-panels-soic.jpg ?
just thought to buy more and chain/arrange display of 12 32x32s  (into 2x6 arrangement) but I am afraid then adafruit HAT wouldn't be sufficient and I would better have at least two 6 panel chains.

maddogs, 2016-08-11:
The board you'd get from OSH Park is just the board. You'd need to buy the level shifters, caps and sockets and solder them all on.

hzeller, 2016-08-14:
12 on a single chain is indeed pushing it, so having the parallel chains is nicer.

OshPark just creates the PCB from the plans, you have to indeed solder it up  yourself.

yarikoptic, 2016-08-15:
Just for lazy like me, if there was a complete recipe "buy this board from here and these components from there" it would be quite useful

hzeller, 2016-08-23:
Currently, I don't have enough resources in my spare time to prepare a kit, but anyone in your local hacker space would be able to help here.

closed 2016-08-23 by hzeller