hzeller/rpi-rgb-led-matrix #194: Adafruit rgb-matrix, half screen shows blue

djdollar, 2016-08-15:
Im having problems with the Adafruit rgb-matrix + raspberry 3 and the 32x32 led matrix.

Half screen(last 16 ros of leds) just shows a static blue while the other half works
[image]

Any ideas of what i could do?

djdollar, 2016-08-15:
if i change some of the settings from:
[image]

to:
[image]

i get this result instead:
[image]

any ideas?

Skid32, 2016-08-18:
i sorta have that problom i looks like this 
[image]

hzeller, 2016-08-18:
Check the cables and that the connectors are properly soldered without shorts.

closed 2016-08-23 by hzeller