hzeller/rpi-rgb-led-matrix #195: I Cant Run Demos

Skid32, 2016-08-17:
Wen I Try To Run A Demo I Just Get This 
[image]

hzeller, 2016-08-17:
This is how it is supposed to look. If you don't see anything on your matrix, it is not connected correctly. Double check your wiring and check out the 'Troubleshooting' section of the README.

Skid32, 2016-08-17:
i can get it to run with the adafruit version

hzeller, 2016-08-17:
Note, the Adafruit version has modified the pinout.
If you are using the Adafruit HAT for instance then this won't work until you do the things mentioned in the relevant section in the README.

Skid32, 2016-08-17:
i fix it sorry for my typeing

closed 2016-08-17 by hzeller