hzeller/rpi-rgb-led-matrix #198: Buy board?

clgyl, 2016-08-18:
Hello!
I wonder if its possible to buy the board (with all components soldered):
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

BR
Clay

mamuesp, 2016-08-18:
Same question from my side ... I googled yesterday, but no success ... and as I have a lot of soldering experience, but nearly none with SMT parts, a version with all SMT parts already mounted would be helpful and appreciated.

depili, 2016-08-18:
Your best bet is probably to seek out the nearest makerspace / hacklab / fablab or similar hobbyist location and ask if anyone there would be willing to assemble the board for you.

clgyl, 2016-08-18:
Thanx for the replies!
I guess I can try to order it from ali express or similar, at least the board which is the critical part for me(since i dont know how to create one). Then I can solder the component on my own.

BR
Clay

closed 2016-08-18 by clgyl