hzeller/rpi-rgb-led-matrix #200: Method to declare panel arrangements for custom transformers as command line arguments

jwheare, 2016-08-23:
e.g. I have 6 64x32 panels arranged in an upside down U chain. It would be great if I could specify that arrangement through command line arguments, to avoid the need to write a custom transformer and then hack led-image-viewer to accept it. I also notice you removed the -L transformer option (though it's still mentioned here: https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use)

Maybe something like a text file that shows the arrangement in ASCII, that can be parsed by the standard command line options. So for my chain of 6 it looks like this:

* First line is dimensions w x h of each panel
* 1 is the closest to the pi, 6 is furthest away

```
64x32
4 3
5 2
6 1
```

hzeller, 2016-08-24:
I have re-added the `-L` now (it was indeed lost in the process of cleanup). Please let me know if things now work again as before (though I guess for you chain of 6 you anyway need some other transformer).

I like the idea for defining the arrangement as text-file (implementation: a transformer that takes a text-file), but didn't get around implementing something like that yet.

But, let's put it this way, I accept pull requests :)

ChazUK, 2016-08-24:
Would it not be better to use an array or object rather than just a plain text file? Possibly add this into the Makefile? This could then include orientation of the screens as well.

BennoG, 2016-09-17:
In my library I use an XML file to discribe the configuration.
So no recompilation needed for what kind of layout you use. (even timing and inversions are configurable from the XML)
Also in this way you can support different PCB layouts without recompileing and differend HUB connections e.g. HUB8 and HUB12 and the HUB75 uesed by this project.
I do use PWM output as soon as the enable is connected to the PWM pin otherwise i use timed enable output.
[hubConfig.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/478151/hubConfig.zip)

Benno

closed 2017-07-30 by hzeller