hzeller/rpi-rgb-led-matrix #201: led-image-viewer interleaves frames of multiple animated gifs

jwheare, 2016-08-23:
Just trying out your new feature here: cdc39384f9910f78d7180a9d0939b17affcdada8

This doesn't quite work as expected for multiple animated gifs. It seems to start interleaving the frames of different images after a few frames of playing the first one, and if the -f flag is left off then the program will terminate at the first point where it would start interleaving.

I can try to get some footage with some test gifs to demonstrate more clearly.

hzeller, 2016-08-23:
Interesting, I probably need to do more testing :)

What do you mean with 'interleaving' ? Does it mix images ? Which would sound odd.

Expected is, if you give it multiple images that it plays them one after another. If you have multiple animated gifs, they will be shown for about 1.5 seconds unless you have given an animation time (`-t`) or the loop count (`-l`). So if you haven't given -t or -l, then the default will kick in, the 1.5 seconds.

(probably it would be better if none of the -l or -t option has been set to default to -l1 instead of -t==-w)

jwheare, 2016-08-23:
Ah yes I think the issue was the unexpected defaults. The images I was using initially didn't make it clear, but after trying with some more obvious test images it's the default 1.5 second timeout that was throwing me off.

`-l1 -f` is the default I would expect, which is how it works if you only specify a single image. But I suppose then you need a way to negate `-f`. `-s` for "single" maybe?

commit 6043d856 (2016-08-24): o Choose a more sensible default for multiple animated
  gifs. Instead of a time-limit (which by default is 1.5 seconds),
  make them loop once.

  Fixes #201
closed 2016-08-24 by hzeller