hzeller/rpi-rgb-led-matrix #202: RPi 2B Adafruit HAT 64x64

ChazUK, 2016-08-23:
Hi,

This isn't exactly an issue, but am I correct in assuming that the only way to get multiple matrixes to work together with the Adafruit HAT is to either use their outdated version of this library, or to hack the IDC connector? Or **can** the Adafruit run multiple matrixes in a chain, but doesn't have the capability to order and orient them?

If the changes where made to allow the Adafruit HAT to run multiple displays in the previous version of this library, then why do any hardware hacks have to be made? Is it impossible to do this in software?

Bare in mind I am a beginner at this kind of thing, first time really touching any electronics, and don't fancy messing up a board if it's still not even guaranteed to work if I do it right.

Thanks.

ChazUK, 2016-08-23:
From quickly testing I'm guessing it's that this library can't chain, and also suffers quite a lot of noise compared to adafruits

adafruit http://imgur.com/P41uVSh
hzeller http://imgur.com/Q6oHgE8

jwheare, 2016-08-23:
What's the full exact command you're running with each library? The command line flags have changed from time to time with this library. Also, what does your `lib/Makefile` diff look like against the latest master branch?

The hardware hack is definitely needed to reduce noise, but it's very quick and simple, especially if you've already managed to solder the HAT correctly. Just take a small piece of wire and connect the holes for pins 4 and 18.

Here's an old shoddy one I did that works great

[image]

hzeller, 2016-08-23:
Chaining should be no problem. You can chain as many panels as you like until you run into refresh rate problems. With the ancient adafruit library or this one - there is no difference.
(I did change the command line flags recently, so what used to be `-c` is called `--led-chain` now).

To multiple chains in _parallel_ , you need to do some different wiring which is probably easier if you use the adapters provided in this project.

Also, do the hardware hack - the noise is something you otherwise can't get controlled.

ChazUK, 2016-08-23:
@jwheare I've already done the flickering hack, although I can't see much difference at the moment.

@hzeller I was more talking about this section https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/README.md#64x64-with-e-line-on-adafruit-hat. For a noobie this reads like you can't chain the matrixes together to make a 64x64 display.

On the Adafruit library I was running the minimal-example demo and `sudo ./led-matrix -L -D 0` with the chaining going 0:TR, 1:TL, 2:BL, 3:BR

I'll take a look at the --led-chain parameter. Is there anyway of setting the order and orientation?

jwheare, 2016-08-23:
That e-line mod isn't necessary with the 32x32 or 64x32 adafruit panels. Looks like you're using 4 32 squares?

I believe the -L flag was removed recently too. Not sure if it's coming back, but you can hard code it back in here:

https://github.com/hzeller/rpi-rgb-led-matrix/blob/5b7f5507dd3009bd674098b6cc59a38a9d343d58/examples-api-use/demo-main.cc#L1144

```
transformer->AddTransformer(new LargeSquare64x64Transformer());
```

Then recompile.

hzeller, 2016-08-24:
Yeah, the -L was accidentally removed in my recent cleanup sweep. Will re-add tonight.

closed 2016-08-24 by hzeller

ChazUK, 2016-08-24:
Thanks!

[image]

Einhornkot, 2017-08-02:
I habe a small question @ChazUK :

How is your final Commando to run the 64*64 matrix?

Philipp :D

ChazUK, 2017-08-02:
@Einhornkot Hey, I haven't actually touched this for about 6 months now, so will have to revisit it.

commit c4f5d64b (2016-08-24): o Re-add a couple of options that got lost in recent cleanup.
  - -L to create 'large' chained 64x64 panel.
  - re-accept some of the old deprecated options so that things
    are not too hard on old-time users.
o Clarify documentation about 64x64 e-line hardware mod.

  Fixes #202
commit b87bf5e2 (2016-08-24): o Re-add some options that got lost in the cleanup
  Fixes #202