hzeller/rpi-rgb-led-matrix #203: Demo running over 2 panels is just duplicated

rodrigosalinas, 2016-08-24:
Hi, I have explained my problem in other issue but maybe it will be more useful as a separated one. I have two P10 16x32 panels chained. I had to write my own Transformer which I have published in the issue #178. With that transformer I can see the demos running on one panel, but duplicated in the second one. You can see it in this video: https://youtu.be/33MPFTckbCQ

Please tell me where I have to work to accomplish to scroll text from one panel to the other. Thank you very very much in advance. And again, thank you @hzeller for you great library.

hzeller, 2016-08-24:
mmh, so it _looks_ like it would look when two panels are chained together physically, but the `-c`- parameter is not given.

However, in your example you _do_ give the parameter.

Maybe something to do with the recent changes ? The `led-matrix` program now actually moved into the sub-directory `example-api-use/` and is called ./demo

https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use

I could imagine that you maybe updated the git, but the old `led-matrix` binary was still in your toplevel directory, but now has slight mismatches with the rest of the library ?

Can you go to the `examples-api-use/` directory and recompile ?
(Or do a fresh `git clone`, then do a fresh compile in the `examples-api-use/` directory; and then copy over your transformer you wrote).

hzeller, 2016-08-24:
Or (I haven't looked in detail yet at your transformer), you might just have a curly 1:8 display, in which you would need to say `-r 8 -c 4`, because each panel is actually internally 'curved around 32 pixel'.

rodrigosalinas, 2016-08-24:
Thank you @hzeller. Running now this command: ```sudo ./examples-api-use/demo -D1 --led-rows=8 --led-chain=4 ./examples-api-use/runtext16.ppm``` I can see the image scrolling both panels except for 4 rows that disappeared. Here you can see it: https://youtu.be/4B1H1ojIbIc

rodrigosalinas, 2016-08-24:
Is it my Transformer?

rodrigosalinas, 2016-08-24:
this is the transformation I do:

    int new_x = 0;
    int new_y = 0;
    if (y < 4 || (y > 7 && y < 12)) {
        new_x = (x * -1) + (24 * ((x / 8) + 1)) - 17;
        new_y = y;
    } else {
        new_x = x + (8 * ((x / 8) + 1));
        if (x < 24) {
            new_y = y - 4;
        } else {
            new_y = (y == 4 || y == 12) ? y : y - 4;
        }
    }

hzeller, 2016-08-24:
First thing I see is that it looks like there are some artifacts probably because the Raspbery Pi 3 is too fast. Can you try `--led-slowdown-gpio=2` ?

Also, make sure that you have enough power connected. These kind of artifacts are as well typical for not enough power. These panels with 1:4 multiplexing use more current than the 1:8 (because they are brighter and meant for out-doors). I would guess you need at least 6A at 5V for this chained display. Make sure that your power supply can provide that and that the cables are not too thin: measure if you have stable 5V directly at the connector of the panel and that it doesn't fluctuate.

And yes, it _could_ be the transformer - I am currently travelling on a mobile phone, so will have a look later. But I first suggest to have a look at the 'physical aspects' described above.

rodrigosalinas, 2016-08-24:
I tried the flag but the problem remains. On the other hand, if it were a matter of power, shouldn't I had trouble before? In the first video you can see all the leds turn on ok. Perhaps is my transformation.

rodrigosalinas, 2016-08-24:
Although the transformation is correct in the sense that it transforms the x, y coordinates called by  the library into coordinates accepted by the panel. I tested one by one with @KentWalker's test-pixel.

rodrigosalinas, 2016-08-24:
Oh, now I got your slowdown advice. My rows missing problem remains but the image quality improved a lot indeed. Thank you very much @hzeller for that.

hzeller, 2016-08-24:
With `--led-rows=16` it uses half the power than with `--led-rows=8`, because it was showing 8 invisible black lines before. It it is very well conceivable that you have power trouble now that you switched to the matching `--led-rows` for the panel.

So you can do `--led-rows=16 --led-chain=4` again to test - it would now do the right scrolling, but you are wasting brightness and refresh rate. If that works, then focusing on the power is a good next step.

rodrigosalinas, 2016-08-24:
That worked indeed. I'll work in the power issue as you point out. What could happen if I just let the panel as is? Thank you very very much @hzeller.

hzeller, 2016-08-24:
If you are using led-rows=16, then you are supplying address lines A, B, C, D; with rows=8 only A, B, C. In panels that have only 8 rows, the connector where D would go is typically shorted to ground. 

So for permanent use make sure that the D-line is not connected as to not accidentally short the digital output to GND and you should be fine.

The refresh rate is lower with rows=16: you can see the refresh rate with `--led-show-refresh` and compare what you get with rows=8 and rows=16.

So nothing 'bad' happens, but the downside is that you loose brightness and that the refresh rate is lower, so it is more likely that people notice flicker.

closed 2016-08-24 by rodrigosalinas