hzeller/rpi-rgb-led-matrix #204: Display not working after long period between demos

ChazUK, 2016-08-24:
I've been playing around with the library now trying to get a crude album art display working. I've found that if I run a few demos and the led-image-viewer a few times, `CTRL+C` to exit and then leave it for a while, then run the demos agin the display does not light up.

Also, after I run my python script (this is just a proof of concept at the moment, please don't shoot me) the display does not light up either.

```#!/usr/bin/python
import websocket
import thread
import time
import json
import spotipy
import urllib
import os

sp = spotipy.Spotify()

def on_message(ws, message):
	obj = json.loads(message)
	if obj["event"] == "track_playback_started" or obj["event"] == "track_playback_resumed":
		track_id = obj["tl_track"]["track"]["uri"]
		track = sp.track(track_id)
		urllib.urlretrieve(track["album"]["images"][2]["url"], "../../utils/images/track.jpeg")
		print "Saving Album Art for " + track["artists"][0]["name"] + " - " + track["name"]
		os.system("sudo ./../../utils/led-image-viewer -L ./../../utils/images/track.jpeg")

def on_error(ws, error):
	print error

def on_close(ws):
	print "### CLOSED ###"

def on_open(ws):
	print "### OPEN ###"

if __name__ == "__main__":
	websocket.enableTrace(True)
	ws = websocket.WebSocketApp(
		"ws://localhost:6680/mopidy/ws",
		on_message = on_message,
		on_error = on_error,
		on_close = on_close)
	ws.on_open = on_open
	ws.run_forever()```

hzeller, 2016-08-24:
mmh. I have not seen this yet. Did you do the steps to make sure that sound is disabled ? The Troubleshooting section has something about it.

Also, how is the program killed ? When you give it a new image, you need to make sure that the previous program is killed. I don't really know python, but it looks like you are starting a new thread and attempt to start a second led-image-viewer when you receive a new websocket message. Then two led-image-viewer run in parallel and fight for the gpio resources and, not seeing anything sounds like a possible outcome.

So always make sure to properly terminate the previous led-image-viewer.

ChazUK, 2016-08-24:
Yep, I'm not killing any processes in python at the moment, it's just a nasty prototype script to see if my idea will work. The sound is not disabled as of yet. I'm grabbing an external USB soundcard tomorrow so that I can figure out a way to disable the onboard sound and feed it to the USB soundcard. Although whilst previously testing, the demo's have always run whilst onboard sound was playing.

hzeller, 2016-08-24:
If you happen to be using the Adafruit HAT and are not using the pwm hardware mod, then it will work also with builtin sound.

So from your description it sounds like the multiple processes are the problem

ChazUK, 2016-08-25:
Thanks @hzeller, It was definitely that the I was using the PWM hardware mod, so the sound was killing the demos. Turned it off whilst waiting for the USB soundcard to arrive and it works.

My crude demo sort of works, but it doesn't unload the image properly. I guess it's probably better to write it in C++ so I can manage the loading of images directly as I can't see that in the python library.

```
#!/usr/bin/python
import websocket
import thread
import time
import json
import spotipy
import urllib
import os
import subprocess

sp = spotipy.Spotify()
process = None

def on_message(ws, message):
	global sp
	global process
	obj = json.loads(message)
	if obj["event"] == "track_playback_started" or obj["event"] == "track_playback_resumed":
		if process is not None:
			process.kill()
		track_id = obj["tl_track"]["track"]["uri"]
		track = sp.track(track_id)
		image = track["album"]["images"][2]["url"].rsplit("/", 1)[1] + ".jpeg"
		print image
		urllib.urlretrieve(track["album"]["images"][2]["url"], "../../utils/images/"+image)
		print "Saving Album Art for " + track["artists"][0]["name"] + " - " + track["name"]
		cmd = ["sudo",  "/home/pi/rpi-rgb-led-matrix/utils/led-image-viewer", "-L", "/home/pi/rpi-rgb-led-matrix/utils/images/" + image]
		process = subprocess.Popen(cmd)

def on_error(ws, error):
	print error

def on_close(ws):
	print "### CLOSED ###"

def on_open(ws):
	print "### OPEN ###"

if __name__ == "__main__":
	websocket.enableTrace(True)
	ws = websocket.WebSocketApp(
		"ws://localhost:6680/mopidy/ws",
		on_message = on_message,
		on_error = on_error,
		on_close = on_close)
	ws.on_open = on_open
	ws.run_forever()
```

Any pointers would be amazing!

closed 2016-08-25 by ChazUK