hzeller/rpi-rgb-led-matrix #206: Chained Displays in Python

chrsvlk, 2016-08-25:
Hi,

I was trying some of the python demos on my 32x64 display (internally chained), but I only see the scrolling text (and other graphics) duplicated on both halves of the display. I tried setting the "-c" argument to 2, but the issue remains. The exact line I run is "sudo python runtext.py -c 2".

In trying to troubleshoot, I made my own script to:

myMatrix = RGBMatrix(32, 2, 1);
print([myMatrix.width, myMatrix.height])

but no matter what I input into RGBMatrix, I only ever see a [32, 32] in my print statement. Am I doing something crazy here?

Thanks

hzeller, 2016-08-25:
This _should_ be working, but maybe the generated Python adapter code has trouble with the recent changes ?  There as a constructor added in the c++ code, maybe that threw off code that looked at particular positions in the binary ?

First step: make sure recompile the python binding from scratch, so that possible binary ABI incompatibilities are excluded.

I am out of town and don't have the network to install python to test myself, but maybe @Saij can have a look ?

hzeller, 2016-08-27:
Uhm, sorry, looks like I messed up the old constructor.
Can you do a 'make clean' and a fresh compile and install of the Python binding to see if things work again ? Sorry about that.

closed 2016-08-27 by hzeller

chrsvlk, 2016-08-27:
Ahh awesome! It works now! Thanks for the quick support, good work!

commit 307ba659 (2016-08-27): o Make original constructor work again (Doh')
  Fixes #206, #207