hzeller/rpi-rgb-led-matrix #207: Probem with python library and 16x32 matrix

haubuchon, 2016-08-26:
Hello 
I have been trying to use your pyhton library with a 16x32 display.

It seems to have a problem (while all the c demos work fine).

for example, if I draw a simple diagonal line from (0,0) to (15,16), the line get broken at (7,7) and continues from the top at the right from (8,0) to (15,7). (I get two short diagonal lines side by side instead of a bigger one going across the whole panel)

In other words, the pixels below the first 8 rows get remapped at 0 instead of being drawn on the bottom half of the display.

Do you have any ideas where I might look to find the problem ?

All the python demo samples show this behavior even when specying -r 16.

Thanks for the help.

hzeller, 2016-08-27:
Uhm, sorry, looks like I messed up the old constructor.
Can you do a 'make clean' and a fresh compile and install of the Python binding to see if things work again ? Sorry about that.

haubuchon, 2016-08-27:
Works perfectly !  Thanks for this great library !

closed 2016-08-27 by haubuchon