hzeller/rpi-rgb-led-matrix #208: Single color panels

gcsuri, 2016-08-28:
Hi Henner,

 Can I use your library with single color panels like this: https://www.aliexpress.com/item/o/614766331.html ? This is 32x16 panel. I would like to make a 64x64 matrix. Would be the best is U chain or two chains?

best regards, Gabor

hzeller, 2016-08-28:
I have not tried these panels myself (but I have ordered some similar), so I don't know how compatible the connector would be (but I assume it is pretty much the same).

(do you mean 32x32 or actual 64x64 with four panels in height ?)

If you have a choice, I would probably go for two parallel chains:

For one, it is much easier to map the pixels, because it just behaves as it is expected, while in the other case, you need to calculate where pixels show up.

Second, if you have an U-shape, you need to turn around some of the panels. These monochrome panels are often directed, light going more down than up for instance (because they are typically mounted high up on a advertisement board), so you would see a brightness difference for the different parts of the display if some  are turned around.

And third, you get a bit more refresh rate and brigtness if you have them parallel, because less data has to be clocked in length-wise (while the parallel clocking doesn't cost anything more for the CPU).

gcsuri, 2016-08-28:
thank you for your quick reply!
So I would like to build a 64x64 pixel matrix from 32x16 panels. So I need to build a 4x2 matrix.
Ok, I can choose 2 chains so I need 4 panels in 1 chain. If I got the panels I will let you know the result.
I think I would need some rotation transformation then :)

BennoG, 2016-09-08:
These panesl use HUB12 and are NOT pin compatible with HUB75
You can however control them with this software if you rewire the 16 pin flat cable.
The HUB12 i have over here have low OE and HUB75 uses high OE
Also the led data signal is inverted so 0=Led on 1=Led off

/*       HUB12      single dual color
      OE  1   2  A
      GND  3   4  B
      GND  5   6  C
      GND  7   8  CLK
     GND  9  10  LAT
     GND  11 12  R
     GND  13 14  G
     GND  15 16  NC
*/
/*       HUB75       RGB
     R1  1   2  G1
     B1  3   4  GND
     R2  5   6  G2
     B2  7   8  GND
       A  9  10  B
       C  11 12  D
    CLK  13 14  LAT
     OE  15 16  GND
*/

serverfire, 2016-10-22:
@BennoG I want to make one of this one color led work , can you help me with that?

BennoG, 2016-10-23:
How are you going to connect the panels to the PI wires or the hzeller pcb ?

serverfire, 2016-10-23:
@BennoG thank you for replay , i have wires them to PI , it is a just one p10 706 one color panel , when i run the pixel-test.cc it divide the panel to 4 sections , it like connecting for panels , when i change the row in code , only one of the that 4 sections works , i have changed everything in code but it did not worked .

BennoG, 2016-11-07:
Sorry for the late reaction, but I don't have time to test the wiring for you at the moment.
It will be at least 1 month before I will have time again. When I have tested it with my test panel i will post the wiring and settings here.

serverfire, 2016-11-07:
@BennoG thank you , i have found a other way for single color led ,using  https://github.com/freetronics/DMD with arduino