hzeller/rpi-rgb-led-matrix #212: outdoor led

serverfire, 2016-08-30:
First of all thank you for great work you have done , you should put a button for donation to this project .

I have problem with python code , i have edited this line of code [led-matrix.cc#L170](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L170)
 
To use my transporter by default and replace it with Null , 

 It work with c++ demo but won't change how python code work and won't transformer it .

Any idea where should put my transformer for python 

Thank you in advance

hzeller, 2016-08-30:
The Python code uses the other constructor, so you have to change the corresponding SetTransformer(NULL) in line 183.

serverfire, 2016-08-31:
@hzeller i'm sure is not calling that function , did not worked , also put a fprintf to make sure . 
maybe @Saij can tell me which file i should edit to set default transformer for python

serverfire, 2016-09-13:
@hzeller 

I have founded a way with last code to make the python to use my transformer but some how there is problem . as you can see in this screen shots it look bad .

graphics.py 

[image]

simple-square.py

[image]

run-text,py

[image]

any idea what should i do ?

hzeller, 2016-09-13:
You didn't mention how you use the transformer ?

hzeller, 2016-09-13:
Maybe your Pi is just too fast for your Matrix ?
In `lib/Makefile`, try commenting out the define for `RGB_SLOWDOWN_GPIO` and set it to 2 or maybe 3.
```
DEFINES+=-DRGB_SLOWDOWN_GPIO=2
```
then, in the Python directory, 'make clean' and then install again.

serverfire, 2016-09-13:
i have set DRGB_SLOWDOWN_GPIO=3 already and have done make clean too . in led-matrix.cc i have put ApplyStaticTransformer(RotateTransformer(180)); in RGBMatrix::RGBMatrix(GPIO *io, int rows, int chained_displays,
                     int parallel_displays)

also in RGBMatrix::RGBMatrix(GPIO *io, const Options &options)

serverfire, 2016-09-13:
I can give you ssh access to rpi and access to ip cam if it is possible for you to take a look , i will pay for your time :)

closed 2017-07-30 by hzeller