hzeller/rpi-rgb-led-matrix #213: 64x64 Display using Python Library

ChazUK, 2016-08-31:
I've been trying to get the 64x64 display working with the Python Library. As far as I can tell none of the demos have this enabled.

I've tried changing /lib/led-matrix.cc Lines 170 and 183 to `SetTransformer(new LargeSquare64x64Transformer());` recompiling the code after the change. Then creating the matrix in Python with `matrix = RGBMatrix(32, 4, 1)` and `matrix = RGBMatrix(64, 4, 1)` but I either get the last 2 panels working, or just the last single panel working with a full picture.

[image]

Not really sure what I'm doing wrong. Has anyone else got this to work for them?

hzeller, 2016-08-31:
Do you use double-buffering by any chance ? Because the transformer set manually in the constructor would only work for setting pixels on the matrix directly not any double-buffering canvases you created, which would be a limitation at this point for this hack for the Python implementation.

`RGBMatrix(32, 4, 1)` sounds like it would be the right setting.

ChazUK, 2016-08-31:
I'm not really sure what you mean by Double Buffering.

hzeller, 2016-08-31:
If you use `CreateFrameCanvas()` somewhere to create an offscreen canvas, then you're using double buffering.

If not, then there might be some other reason why it won't work (can't think of any right now), but then you have to probably just do the pixel mapping manually in Python for now (so determine the actual pixel location before calling SetPixel()).

I try to get this more simple in Python, but there is only limited weekend time :)

ChazUK, 2016-08-31:
This is my current Python script. I'm only using `matrix = RGBMatrix(32, 4, 1)` as far as I'm aware.

```#!/usr/bin/python
import websocket
import thread
import time
import json
import spotipy
import urllib
import cStringIO

from rgbmatrix import RGBMatrix
from PIL import Image

sp = spotipy.Spotify()
prev_track_id = ""
prev_album_art = ""
matrix = RGBMatrix(32, 4, 1)

def on_message(ws, message):
    global sp
    global prev_track_id
    global prev_album_art
    global matrix

    obj = json.loads(message)

    if obj["event"] == "track_playback_started" or obj["event"] == "track_playback_resumed":
        track_id = obj["tl_track"]["track"]["uri"]

        if prev_track_id != track_id:
            prev_track_id = track_id
            track = sp.track(track_id)
            album_art = track["album"]["images"][2]["url"];

            if prev_album_art != album_art:
                prev_album_art = album_art
                print "Loading New Album Art for " + track["artists"][0]["name"] + " - " + track["name"]
                file = cStringIO.StringIO(urllib.urlopen(album_art).read())
                image = Image.open(file)
                image.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)
                matrix.SetImage(image.convert('RGB'))

def on_error(ws, error):
    print error

def on_close(ws):
    print "### CLOSED ###"

def on_open(ws):
    print "### OPEN ###"

if __name__ == "__main__":
    websocket.enableTrace(True)
    ws = websocket.WebSocketApp(
        "ws://localhost:6680/mopidy/ws",
        on_message = on_message,
        on_error = on_error,
        on_close = on_close)
    ws.on_open = on_open
    ws.run_forever()```

hzeller, 2016-08-31:
Ok, this looks pretty good.
If you do get if you print the matrix dimensions ?

```
print("w=%d,h=%d" % (matrix.width, matrix.height))
```

ChazUK, 2016-08-31:
`print 'width={}, height={}'.format(matrix.width, matrix.height)` produces `width=128, height=32`

hzeller, 2016-09-01:
I guess something went wrong with recompiling the python library - I just tried things and it works exactly as it should.

1) edit `lib/led-matrix.cc` line 183 and change `SetTransformer(NULL);` to `SetTransformer(new LargeSquare64x64Transformer());`

2) Fresh compile of everything. In toplevel directory
```
   make -C lib clean all
   sudo make -C python clean
   make build-python
   sudo make install-python
```
3) Test:
```
$ sudo python
Python 2.7.9 (default, Mar  8 2015, 00:52:26) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from rgbmatrix import RGBMatrix
>>> matrix = RGBMatrix(32, 4, 1)
>>> print ('width={}, height={}'.format(matrix.width, matrix.height))
width=64, height=64
>>> 
```

ChazUK, 2016-09-01:
This has corrected the issue. My apologies as I didn't know you had to clean make the lib/ folder.

Would it be possible to have the transformer accessible in the python library, or even move it out to the Makefile as a default Transformer.

hzeller, 2016-09-01:
I am working towards having this switchable from Python (this is issue #101). I try to avoid more switches in the Makefile, because it requires re-compilation, which makes it more tedious to change something.

ChazUK, 2016-09-01:
Makes sense!

Thanks for the help.

closed 2016-09-01 by ChazUK