hzeller/rpi-rgb-led-matrix #215: Component for active-3 adapter 5V pads

jwheare, 2016-09-08:
I took the leap with your custom boards and ordered parts and 2 boards (one of which I already fried 😞 ) but I was wondering what component you had in mind for the 5V/GND power input pads. It doesn't look like a barrel jack would fit there.

Also, because it wasn't mentioned in the bulleted BOM I completely forgot about it when shopping for all the other parts. Grateful for any guidance you can provide.

maddogs, 2016-09-08:
If you need another active board (bare), send me a mail as I have a few spare.

hzeller, 2016-09-08:
For the 5V supply, I don't have any connector in mind; this is mostly meant as bare soldering pads (if you want to power the Pi from there).

In a typical scenario, you have the Panels powered by 5V anyway, so you can wire the Pi power to it easily that way without having to go through the hassle of wiring up a MicroUSB.

jwheare, 2016-09-08:
Ah ok. I was hoping to use some 5V barrel jack wall warts I have lying around instead of hacking more into my ATX power supplies. I probably have some adaptors I can sacrifice though, otherwise I'll just stick to USB.

@maddogs I'll be in touch, thanks!

closed 2016-09-08 by jwheare