hzeller/rpi-rgb-led-matrix #217: Glitches with active-3 adapter, chain=4, parallel=3

jwheare, 2016-09-10:
Any idea what might be causing this glitching, particularly in the upper panels?

https://youtu.be/NzYrocDpumY

I've ruled out the power supply, panels, IDC and power cables so far. Will test with another Pi tomorrow.

I tried slowdown-gpio all the way up to 4 and it helps a bit but doesn't eliminate it.

hzeller, 2016-09-10:
How beefy are your cables ? Your display will probably need around 12*3A = 36A, and if you supply that with a common cable, then some panels 'further away' can be left in the rain.

In particular the cables that come with the panels are often underdimensioned.

Measure the of each voltage _directly_ at the power connector of the matrix to see how much voltage you get there. It should not drop below 4.8 or so (some power supplies allows to slightly increase the voltage, say 5.25V to have more wiggle room).

The cables also could not be able to supply the current peaks needed due to inductances of the cables. I would solder-on Low ESR capacitors directly where the matrix connectors are soldered on the board.

Make sure to read the "Word about power" section in the documentation. https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power

jwheare, 2016-09-11:
I'm using an ATX power supply that can drive 40A on the 5V rail, and the same cables I was previously using with an adafruit hat and one long U-shaped chain from a single socket. There was a little occasional glitching on a few pixels with that setup, which is what prompted me to try to the active-3, but nowhere near this level.

I'll take some measurements tomorrow to be sure. Is there a chance this could be caused by a soldering error or damage to the board?

hzeller, 2016-09-11:
If you used the U-shaped, then your display was darker, so it is very likely that you now exceed the power.

Soldering or damage to the board is unlikely. I suspect power.

BennoG, 2016-09-11:
I have 150A supply for 20 panels  32x16 pixels half the size you have.
They each draw 8A if all leds are on in white.
Also use short data cables between the panels < 20cm
Don't know the exact display type you have but 6 times 8A is 48A so power could be an issue.

jwheare, 2016-09-11:
I rechecked my power situation, ran 2 extra dedicated cables from the power supply instead of double ended connectors, and also shortened the IDC cables from the adapter to the first screen, then set slowdown-gpio=4 and it's now running smoothly, no extra power needed. Thanks for the advice.

closed 2016-09-11 by jwheare