hzeller/rpi-rgb-led-matrix #218: Getting sound out of Pi

currentlysober, 2016-09-13:
Dear all, is there any way to get sound out of the Pi? To use some other hardware sub-system?
There are 2 ways to get sound: via PWM pins or via HDMI. Both ways are disabled now, aren't they?
I was thinking about showing text on the screen and playing music in the same time.
I'll be grateful for any help.
Regards!

hzeller, 2016-09-13:
Yes, I have heard from people doing that, using external USB sound adapters. Essentially you only need to make sure that the snd module that is using PWM is not loaded (the suggestion in the readme is a bit more radical, but simpler to deal with).

currentlysober, 2016-09-14:
ok, thank you!

ChazUK, 2016-09-16:
I purchased a £5 USB audio adapter from Amazon and it worked a treat. It also drastically increases the audio quality

closed 2017-07-30 by hzeller