hzeller/rpi-rgb-led-matrix #219: adafruit Hat flickering (soldering 4 and 18 and uncommenting does not work)

jasieksta, 2016-09-14:
Hi i have a problem with flickering image on my rgb matrix. When trying to do actions described in Makefile, after recomiling matrix entirely stops displaying images. Before uncommenting it works perfectly fine (only with flickering). I am using raspberry pi 3 model B

hzeller, 2016-09-14:
did you also do the necessary hardware mod?

jasieksta, 2016-09-14:
Yes, of Course. I soldered  4 and 18 pinholes together

hzeller, 2016-09-14:
Mmh, strange, this should work. And with your second compile you also made sure to give `HARDWARE_DESC=adafruit-hat` (which you have to do in any case, PWM or not) ?

jasieksta, 2016-09-14:
Yes, here are my steps:
1) connected pinholes
2)uncomment 
3) make clean in main folder.
4) make in main folder. 
5) make in utils folder
6) reboot
7) led-image viewer does not work anymore

hzeller, 2016-09-14:
To confirm, in step (4), did you prefix `HARDWARE_DESC=adafruit-hat` to your make ?

Also, if you switch on PWM, you need to make sure to disable the sound-system, did you read that section in the README ? (Troubleshooting section).

jasieksta, 2016-09-16:
i checked all this and apparently i did not disable the sound system. Now it works great! Thanks for your help, your library is amazing and sorry for the inconvenience :)

closed 2016-09-17 by hzeller