hzeller/rpi-rgb-led-matrix #220: Best way to include RPI RGB Library with my project

bryanmcgrane, 2016-09-15:
I've been working on a small home project written in C++ which connects to many services using the RPI3 and plots information onto a 16x32 grid. I originally wrote my code in Python and had great success interfacing with the python libs, but I decided to move C++ in an attempt to speed up the performance. While my code is running much faster in C++, I am stuck trying to properly integrate and compile this library with my current project. I only have a beginners grasp on C++ and have just become comfortable using CMAKE to build on my home machine (ubuntu) and the pi for testing. I'm also somewhat unfamiliar with how to link my project with github, so I've just been downloading the libraries and adding them to my project directory.

I am currently using the Adafruit hat, but I'll be designing my own pcb off your examples. I'm also using a single 16x32 panel, so I feel like this will be easy.

After reading through the documentation, here is how I would like to implement this library in my main code. Also note I would not like to use command line arguments:

```
int main{
using namespace rgb_matrix;
RGBMatrix::Options matrix_options;
rgb_matrix::RuntimeOptions runtime_opt;

matrix_options.rows = 16;
matrix_options.chain_length = 1;
matrix_options.parallel = 1;
matrix_options.pwm_bits = 11; // for now
matrix_options.pwm_lsb_nanoseconds = 200; // for now
matrix_options.parallel = brightness = 100;
matrix_options.disable_hardware_pulsing = false;

runtime_opt.daemon = 0; // Don't want daemon for now

RGBMatrix *matrix = CreateMatrixFromOptions(matrix_options, runtime_opt);

if (matrix == NULL)
 return 1;

// Testing the canvas
Canvas *canvas = matrix;
canvas->Clear();
// wait command here
canvas->Fill(255, 0, 0) // fill red
// wait command here
canvas->SetPixel(0, 0, 0, 0, 255) // set pixel 0,0 to blue
// wait command here
canvas->Clear();
```
Before I dive into this fully, I just want to know I'm on the right track here. Does all of the above seem correct? What would be your best recommendation for linking your code with my project? Thanks again for your time!

hzeller, 2016-09-15:
This looks good (the above won't compile of course because main() is not correctly declared and there is no closing bracket, but I guess this was more for illustrative purposes).

Recommendations about using and linking this project you find in the readme in the `examples-api-use/` directory:
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#integrating-in-your-own-application

Let me know if there are things I should improve in that doc.

hzeller, 2016-09-15:
```
matrix_options.parallel = brightness = 100;
```
looks like you didn't mean that :)

bryanmcgrane, 2016-09-15:
Oops good catch on my bugs. I'll fix that in case someone else reads it. I'll check through your docs and post here if I need further clarification. Thanks!

bryanmcgrane, 2016-09-16:
So I have looked at the documentation and am now trying to add your project to my CMakeLists.txt file. 

Here is what I'm starting with - I recursively find all my .cpp and .h files for my project in order to compile. How would you recommend I integrate your make suggestions into the below file?

```
cmake_minimum_required(VERSION 2.8 FATAL_ERROR)
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
project(base_station)

include_directories(${PROJECT_SOURCE_DIR}/Libs)
include_directories(${PROJECT_SOURCE_DIR}/Include)

file(GLOB_RECURSE SRC_FILES ${PROJECT_SOURCE_DIR}/Libs/*.cpp)
file(GLOB_RECURSE HEADER_FILES ${PROJECT_SOURCE_DIR}/Include/*.h)

message(${SRC_FILES})
message(${HEADER_FILES})

add_executable(base_station base_station.cpp ${SRC_FILES} ${HEADER_FILES})

target_link_libraries(base_station -lpthread curl)
```

My file structure looks like this:
(root)
base_station.cpp (main code)
    Libs (folder)
        a.cpp
        b.cpp
        c.cpp
    Include (folder)
        a.h
        b.h
        c.h
    matrix (your library)

Sorry again for all the questions, I am new to using make files!

hzeller, 2016-09-16:
Sorry I can't really help here. I don't know anything about cmake, I usually just use simple Makefiles.

(CMake only really makes sense for large projects with many directories and files and that should be platform independent etc.
For everything else with a handful of c++ files, using Makefiles is much easier to deal with. The learning curve for cmake is much more steep (because you need to know how make works and then understand what cmake does around it))

Maybe someone else subscribed to updates of this project can help ?

bryanmcgrane, 2016-09-16:
I'd love to write this as a simple makefile. I'm happy to take that route if it gets me to the end goal. 

How would you suggest I write a simple makefile for this project given the above file structure?

hzeller, 2016-09-16:
Start reading the makefiles in `example-api-use/` and `utils/` and understand what all the variables mean such as `$<`.

A good starting point is reading the documentation of gnu make:
https://www.gnu.org/software/make/manual/html_node/index.html

bryanmcgrane, 2016-09-16:
I've got it using cmake! For those who need to use cmake moving forward, here is my CMakeLists.txt file. I'm sure there is a lot of room for improvement, but it compiles and runs great for now:

My file structure looks like this:
(root)
-base_station.cpp (main code)
-Libs (folder)
--a.cpp
--b.cpp
--c.cpp
-Include (folder)
--a.h
--b.h
--c.h
-matrix (your library)

CMakeLists.txt
```
cmake_minimum_required(VERSION 2.8 FATAL_ERROR)
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
project(your_project_name_here)

set(HW_FOLDER adafruit-hat) #can be adafruit-hat, classic, or regular

include_directories(${PROJECT_SOURCE_DIR}/Libs)
include_directories(${PROJECT_SOURCE_DIR}/Include)
include_directories(${PROJECT_SOURCE_DIR}/matrix/lib)
include_directories(${PROJECT_SOURCE_DIR}/matrix/include)
include_directories(${PROJECT_SOURCE_DIR}/matrix/lib/$(HW_FOLDER))

file(GLOB_RECURSE SRC_FILES ${PROJECT_SOURCE_DIR}/Libs/*.cpp)
file(GLOB_RECURSE HEADER_FILES ${PROJECT_SOURCE_DIR}/Include/*.h)
file(GLOB_RECURSE RGB_SRC_FILES ${PROJECT_SOURCE_DIR}/matrix/lib/*.cc)
file(GLOB_RECURSE RGB_HEADER_FILES ${PROJECT_SOURCE_DIR}/matrix/include/*.h)
file(GLOB RECURSE HW_HEADER_FILE ${PROJECT_SOURCE_DIR}/matrix/lib/$(HW_FOLDER)/*.h)

add_executable(your_project_name_here your_project_name_here.cpp ${SRC_FILES} ${HEADER_FILES} /
${RGB_SRC_FILES} ${RGB_HEADER_FILES} ${HW_HEADER_FILE})

target_link_libraries(your_project_name_here -lpthread curl) # I need pthread and curl for my project
```

closed 2016-09-16 by bryanmcgrane

K0rkunc, 2021-08-20:
> I've got it using cmake! For those who need to use cmake moving forward, here is my CMakeLists.txt file. I unfortunately had to move the led-panel-pin-mapping.h file to the matrix/include directory. I'm sure there is a lot of room for improvement, but it compiles and runs great for now:
> 
> My file structure looks like this:
> (root)
> -base_station.cpp (main code)
> -Libs (folder)
> --a.cpp
> --b.cpp
> --c.cpp
> -Include (folder)
> --a.h
> --b.h
> --c.h
> -matrix (your library)
> 
> CMakeLists.txt
> 
> ```
> cmake_minimum_required(VERSION 2.8 FATAL_ERROR)
> set(CMAKE_CXX_STANDARD 11)
> set(CMAKE_CXX_STANDARD_REQUIRED ON)
> project(base_station)
> 
> include_directories(${PROJECT_SOURCE_DIR}/Libs)
> include_directories(${PROJECT_SOURCE_DIR}/Include)
> include_directories(${PROJECT_SOURCE_DIR}/matrix/lib)
> include_directories(${PROJECT_SOURCE_DIR}/matrix/include)
> 
> file(GLOB_RECURSE SRC_FILES ${PROJECT_SOURCE_DIR}/Libs/*.cpp)
> file(GLOB_RECURSE HEADER_FILES ${PROJECT_SOURCE_DIR}/Include/*.h)
> file(GLOB_RECURSE RGB_SRC_FILES ${PROJECT_SOURCE_DIR}/matrix/lib/*.cc)
> file(GLOB_RECURSE RGB_HEADER_FILES ${PROJECT_SOURCE_DIR}/matrix/include/*.h)
> 
> add_executable(base_station base_station.cpp ${HW_HEADER_FILE} ${RGB_SRC_FILES} /
> ${RGB_HEADER_FILES} ${SRC_FILES} ${HEADER_FILES})
> 
> target_link_libraries(base_station -lpthread curl)
> ```

i wont work that framebuffer /home/pi/new_kickass/matrix/lib/framebuffer.cc:392: undefined reference to `matrix_hardware_mappings'

meirg, 2022-04-27:
> i wont work that framebuffer /home/pi/new_kickass/matrix/lib/framebuffer.cc:392: undefined reference to `matrix_hardware_mappings'

Change `file(GLOB_RECURSE RGB_SRC_FILES ${PROJECT_SOURCE_DIR}/matrix/lib/*.cc)` to  `file(GLOB_RECURSE RGB_SRC_FILES ${PROJECT_SOURCE_DIR}/matrix/lib/*.cc?)` to also compile the c file