hzeller/rpi-rgb-led-matrix #221: DrawText not working on rc.local boot?

tehmooo, 2016-09-17:
Bit of a weird one... Written a python script to turn the LED panel into a clock, and it all works well and great when i run it from terminal - but if i add it to rc.local and boot it with the pi the DrawText stuff doesn't appear. I get lines from DrawLine and I can fill sections and manually setPixels... but DrawText isn't appearing. Any ideas?

BennoG, 2016-09-17:
something with the fonts that can not be found ?
just my 2c
Benno

hzeller, 2016-09-17:
Make sure that you set an absolute path the font, because your binary runs in some working directory depending on where the rc.local started it.

tehmooo, 2016-09-17:
Aha! That's the problem, brilliant. Thanks for the help!

closed 2016-09-17 by tehmooo