hzeller/rpi-rgb-led-matrix #222: New panel has darker colors compared to other ones

Calabraun, 2016-09-19:
So last year I built an LED wall consisting of 9 64x32 6mm led matrix boards. The great idea about these boards is that you can order a new board for just the one board that goes bad and not have to replace all of them. Well, I guess the problem with that thinking is they constantly update the boards and they differ slightly from the ones you got before!

I just ordered a new panel from Adafruit when I accidentally broke on of the chips off one of my originals. The brightness is noticeably darker than the other panels, however :( It has smaller chips on the back (ICN2028AP) than the chips on the original boards (JX15020GF). I asked Adafruit if they had any of the old batch and they do not.

So..... Is it possible to change the brightness of just the 1 board so that it matches the other ones? I see there is this command...

--led-brightness=<percent>

I guess I would need something like this if i were to want to change the brightness for just panel 3 chain 1...

--led-brightness-specific-panel=3,1,80

which lines of code should I be looking at for a modification like this?

I normally run this wall with the following options:

led-matrix -r 32 -c 6 -P 3 -D 0

Photo of issue (new panel is top right) https://ibin.co/2veypt5GbqW8.jpg

hzeller, 2016-09-20:
So, right now, there is no thing that specifically targets one panel regarding brightness, though having something that adapts (probably per pixel) would be good to have (similar to issue #193).

It looks like you actually need to tone down all the panels while leaving the new on 100%, as the new one is darker.

For now, I'd probably just manually hack this here:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/cc96e209c645dc53e3bafdaf8d89a3ecda8dfeb8/lib/framebuffer.cc#L309

And add something like
  brightness_ = (x >= 128 && y < 32) ? 100 : 90;

the condition needs to be adapted to what identifies that panel and puts the brightness to 100%, otherwise put it to 90%. Not pretty, but should get you up and running for now.

Calabraun, 2016-09-20:
Well I feel pretty good that I was in the right area of code, though I created separate independent offset variables for r, g, and b within the SetPixel method (because it seems that the brightness is a bit different of an offset for each of the individual colors, blue appears to have the biggest difference between the new panel and original panels). I am still having issues getting it just right.

I also thought about soldering that chip back onto the old board, though most of the solder pads were ripped off along with the chip too, so a pretty monumental task. Or try to find another older 64x32 6mm panel with the same chips as mine on ebay / chinese site.

For now, I will just use the new panel I have and go with your line of code, it's much simpler than my code and actually gets pretty darn close. 

Thanks so much!

chrisoutwright, 2017-04-24:
Is this hack (brightness_ = (x >= 128 && y < 32) ? 100 : 90;)  working as for now?
Could an inline option be created for such a hack? I tried the above but I do not seem to get anything different.

chrisoutwright, 2017-04-24:
I have tested some color calibration and used two old panels in the top and two new in the bottom. With image editor I created a background which resulted in circa equal white.

[image]

Could one do the coding to transform the color mapping with the offsets? How would I proceed?

hzeller, 2017-04-24:
If you have issues not only with the brightness but with the color mix, you need to apply the brigthness hack for each individual r/g/b component (then probably directly in MapColors(), but you need to give it the coordinate for the decision; little bit of hacking needed).

The shift registers are fine, and possibly the current source the use internally as well. The problem might be just manufacturing differences of the batch of LEDs they use. Always get all the LED panels you need for one installation at once.

Also make sure that all your panels have a solid connection to power - too thin cables can make the voltage drop. Light up the panel full white and measure each power input at the panels, if they are at 5V. If the voltage drops too far, the colors shift towards red, as the blue LEDs cannot be lit anymore fully.

chrisoutwright, 2017-04-24:
Next time I will definitely buy in big bulk orders, I just did not think that the color coefficient correction needed would be that dramatic. 
Maybe the panel has been used already and the blue has faded out a bit, or it is just manufacturing. 
I will try the `MapColors() `route this week, if my ability allows it, but i guess I am not the first to encounter such an issue. 
I wonder how peole replace defect panels and cope with the color differences at a hobby level like this alternatively. 
Are there already RGB Led matrix panels with a second set of RGB leds and a circuit for color coefficient correction? Some patents are already out i think but nowhere to buy such a matrix.

hzeller, 2017-04-24:
* I suspect that these are just different batches of LEDs, not necessarily worn-out used LEDs. But it is conceivable as the Chinese economy is much better in environment-friendly recycling.
* You could try to find the circuit that provides the constant current on the board and reconfigure that (probably a set of resistors if you're lucky). However, often it is hard to find any datasheets for the chips used on these panels, so it might be hard.
* reproduction and perception of color is complicated, since the eye can quite adapt to the baseline of what 'white' is. A white piece of paper looks white in the sun and under incandescent light. It all depends on the context; keyword white-balance. The main problem is if you put more yellow-white and blueish-white next to each other, then it is pretty noticeable. So the tint change is actual fact of different mixture of Red/Green and Blue. The perception can keep up to still perceive 'white' in these cases, but this breaks down of course when they are side by side.

closed 2018-02-17 by hzeller