hzeller/rpi-rgb-led-matrix #225: Difficulty in printing two rows

Gadag007, 2016-09-27:
Hi, 
I am working on led matrix 64x32 with Rpi3, When I trying to print runtext16.ppm with command of sudo ./demo -D 1 -c 2 -r 16 runtext16.ppm, it spiting the words , some of the words printing top 8 rows and reset of the words printing from 16th rows , can any one please suggest the solution for this ? also i want to know how the ppm works when we create ppm file image, will take full 32 rows or can be spited as per over requirement and how can i run runtext16.ppm in two rows with different speed?

Thanks

hzeller, 2016-09-27:
If you have a 64x32, your options should be -c2 -r32

Gadag007, 2016-09-27:
I tried with that , how can I use  two runtext16.ppm at time in 32 rows display and also how can i run with different speed ?

Thanks

hzeller, 2016-09-27:
well, you'd need to write your own program to do that.

Gadag007, 2016-09-27:
when we create ppm file or run , weather it take hole 32 rows ? can't make use of rest of the rows to display static words to display at time ?

hzeller, 2016-09-27:
the PPM scrolling is really only an example. But you can easily write a program where you use part of the screen to display some scrolling image and another to display static stuff.

Or you generate an animated gif and use the led-image-viewer in utils/

Gadag007, 2016-09-27:
Ok thanks for valuable time , I will get back you if i found any query.

Thanks

closed 2016-10-13 by hzeller