hzeller/rpi-rgb-led-matrix #226: Problem running example

jhchong92, 2016-09-29:
Hi sir, really appreciate the great work and support you've been giving.

I encountered a problem when running the demo.
What i did was:-
`make` in examples-api-use directory
`sudo ./demo -D0` just halts the terminal and I can't use Cntrl-C to exit.
The LEDs do not light up as well.

 what could be causing this?

jhchong92, 2016-09-29:
Hey... I was going through the troubleshooting guide... I added the command line --led-no-hardware-pulse and it's working now.

Helbrax, 2016-09-29:
Try reading the "Bad interaction with Sound" in the help/readme.  The new raspian os has some weird interactivity with sound that older versions didn't seem to have.  Disabling the sound drivers prevents them from screwing around with the hardware subsystem, which just so happens needs to be used by the led library.  That could explain why when you toggle hardware pulses off, suddenly it works, since it's no longer using the conflicting hardware.

jhchong92, 2016-09-30:
Hey, thanks for the help. Does that mean if I need the sound drivers I have to downgrade to Rpi2? There are no conflicting hardware issues in Rpi2 right?

Helbrax, 2016-09-30:
There are still conflicts.   On my rpi2,  even though I can run the led matrix without disabling the sound drivers, there is no sound.  I get a high pitch whining and buzzing noise constantly,  which I imagine is the led pulses going through the hardware.   There is another issue open where someone was able to enable certain pieces of the sound modules and get an external USB sound system to work.

jhchong92, 2016-09-30:
Great! Thank you so much for your info.

hzeller, 2016-09-30:
Yep, what @Helbrax said: independent of the Raspberry Pi version, if you want to use the hardware pulses, the library needs the subsystem in the SoC that also would be used for the PWM sound generation.
But you can use some external USB sound device and they will work in parallel (also, _any_ cheap USB soundcard is better in sound quality than what the Raspberry Pi can provide natively).

jhchong92, 2016-09-30:
Hah, glad to know of the usb alternative :)

closed 2016-09-30 by jhchong92