hzeller/rpi-rgb-led-matrix #227: Flickers with P10 32x16 panels

marcinzdunek, 2016-10-02:
Hello,

Few days ago I received some Chinese 32x16 P10 panels. They are quite big and uses significant amount of power. I have managed to run them using these parametres:
led chain - 4 for each panel
rows - 16
I was forced to write a new transformer, really strange one (I can upload photo or code if someone would like to take a look).
Everything was ok, till the time.
When displaying image (I got 64x64 matrix using 8 panels connected into two chains) it sometimes diplays 'ghost pixels' in random locations. These pixels have like 30,10,15 RGB color (randomly low values). 
Another issue is short-time drop of brightness level. From time to time (average 2 times / 10 secs. randomly) screen dims for less than 1 second. 

Before I managed to run yours lib on another panels (for example 32x32 P6) and had no problems like that, everything was fine.

BennoG, 2016-10-03:
Be sure you have a big power supply, I use a single 40A power supply for every 4 panels.
so for 20 panels you need 5 times a 40A power supply.

hzeller, 2016-10-03:
Yes, what @BennoG said. These P10 panels are meant for outdoor, so they might draw quite a lot more current than smaller panels.

Measure the voltage directly at the panels and see if it does not drop too much when displaying a white image. If there is too much difference between power supply and panel (Personally, I try to keep that below 0.4V) then the cabeling is too thin. If there is drop on the cables, sometimes it is sufficient to slightly increase the voltage of the power supplies to say 5.25V (but not too much: the voltage drop is only there if the panels draw a lot of power, if they don't they might be disturbed by the over-voltage).

If the voltage already drops at the power supply, then your supply is not up for the task. Measure the current to see how much your total assembly draws (for these kind of currents, you want a clamp meter; even cheap ones such as the Uni-T UT210E are sufficient). Then get some power supply that can do that plus some margin.

If you use multiple supplies, make sure to connect all grounds together.

[Beefy cabeling and maybe some larger capacitors directly at the power terminals of each panel can help](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power)

@marcinzdunek if you would like to provide a pull request with your transformer, that might help some people that have similar panels.

marcinzdunek, 2016-10-03:
Okay thanks a lot, I will check power and write a comment asap.

marcinzdunek, 2016-10-04:
Unfortunately power has nothing to do with this issue. I have just noticed, that when brightness drops, my framerate drops too, from 900 Hz to 500 Hz. Any other ideas? :(

hzeller, 2016-10-05:
Do you have some other process running on your system that bogs it down ?
Maybe your Pi is getting too warm and it throttles the CPU frequency ?
900 Hz update rate sounds like  you have tweaked the PWM bits or nanoseconds - if that core that is doing the update constantly and does not get enough idle time, the OS is throtteling that, so you might relax some of these timings.
In any case, you should make sure to reserve the update core if you haven't already (the isolcpus thing mentioned in the README).

hzeller, 2016-10-07:
Also, I presume you switched off the sound system ?

closed 2016-10-13 by hzeller

serverfire, 2016-10-14:
@marcinzdunek can you upload your transformer ? it would be useful for me , thank you