hzeller/rpi-rgb-led-matrix #228: Shifting runtext16.ppm with different potions

Gadag007, 2016-10-04:
Hi, 
I am using hub75(32x32),1:16 Multiplex , I am trying to shift reuntext16.ppm with different potions, but i am not able to do that , please any one suggest me , How it can be done and  where i can modify the code.

Thanks

hzeller, 2016-10-04:
You are not very descriptive what you need, but [this is the place](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/demo-main.cc#L281) where the image scroller example does its things.

However, this is really a simple hacky example and not really a good place to add extensions. You should really just read that code, get inspired and write a fresh program what you'd like to do. You can get inspired by the [led-image-viewer](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/utils/led-image-viewer.cc) or the python example [scrolling an image](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/python/samples/image-scroller.py).

Gadag007, 2016-10-04:
Hi hzeller, I have updated my query.

Gadag007, 2016-10-04:
Hi Maddogs, Thanks for your response, But I'm working on C++ APIs.

hzeller, 2016-10-04:
So if you want to shift the image vertically, why don't you just add a constant (e.g. 16) to `y` when call `SetPixel()` @Gadag007 ?

Gadag007, 2016-10-05:
Thanks so much hzeller, you rock... I was able to achieve the shift steps. 

Also, I had to spawn a new thread with imageScroller class replicated to make parallel ppm at once. I will make it more readable and post it below so that it can help someone else.

Lastly, I wanted to run the ppm and stop it once it hits the left most column in the matrix, probably by changing scroll_ms to 0. I could go with text-example but, I want to achieve this in ppm.

 Any suggestions?

Thanks again.

hzeller, 2016-10-05:
Modifying scroll_ms sounds like a particular hacky way to do that. All you need to do is not continue scrolling once your end-condition is met.

closed 2016-10-13 by hzeller