hzeller/rpi-rgb-led-matrix #229: Low brightness on colors other than 1 bit

marcinzdunek, 2016-10-05:
Hello,

When displaying white color on different panels I got great brightness, it's lighter than producer says (for example 7500 instead of 7000). However when I am trying to display filled panels (5 panels x 2 chains) using amber color (255,167,5) I got only about 34% of maximum brightness, where this maximum should be 68% (using luminance calculation formula). What can we do in the library to increase brightness of other colors?

hzeller, 2016-10-05:
You didn't say how you got the 34% and 68% figures.

The input color tuple is sRGB, so they have a roughly exponential response because your eye perceives things logaritmically, So 34% sounds about the right expected brightness if you measure it linearly.

(you can switch off the luminance correction in the matrix `set_lumiinance_correct(false)`, but then you notice that the colors don't look right).

So it sounds like everything is working as expected.

marcinzdunek, 2016-10-06:
But we have measured luminance using professional aprature, and it showed to us that brightness of amber color is about 2380 where producer says it should be 68% of 7000 = 4760.

hzeller, 2016-10-06:
If you are measing luminance, you are measuring the _linear_ value.

Your eye responds logarithmically. The library corrects for that.
So you expect a half bright (say (128,128,128)) much less than 50% luminance, but it looks half bright to the eye.

closed 2016-10-13 by hzeller