hzeller/rpi-rgb-led-matrix #231: Compile error

dennys78, 2016-10-09:
hi

when I "make led-image-viewer" command I receive this message :

/usr/bin/ld: warning: libjpeg.so.8, needed by /usr/lib/libGraphicsMagick.so, may conflict with libjpeg.so.62

Apparently led-image-viewer works but any GIF is show on the panel

please help me

hzeller, 2016-10-10:
This is only a warning, so things should work fine.

Your sentence is not quite clear: you tried and show images such as PNG or JPEG but then GIF didn't work ?

dennys78, 2016-10-10:
It does not work any images... my procedure is this:
# git clone https://github.com/hzeller/rpi-rgb-led-matrix.git
edit Makefile and correct in /lib HARDWARE_DESC?=adafruit-hat 
uncomment DEFINES+=-DADAFRUIT_RGBMATRIX_HAT_PWM
# cd ..
# make (compile but not appare executable led-matrix) 
# cd utils 
# make led-image-viewer
when lunch  #./led-image-viewer -c1  pesci.gif  
appear :
Load images...
Display.
But nothing on the panel :( :( :(

hzeller, 2016-10-10:
Did you confirm that the regular demos work with your Adafruit PWM hardware modification ?

dennys78, 2016-10-10:
no don't work

hzeller, 2016-10-10:
Then you first need to make sure that that is working. The README has a lot of information regarding troubleshooting the Adafruit modifications.
For instance, did you disable the sound system ?

dennys78, 2016-10-10:
I will try with this command :

./led-image-viewer --led-no-hardware-pulse -C -c1 pesci.gif

but nothing

dennys78, 2016-10-10:
It's possible send me a version correctly compiled for raspberry pi 3 ??!? Or use another system?!!

hzeller, 2016-10-10:
This has nothing to do with the Raspberry Pi 3 - the library compiles fine there and works. You have something in your electrical set-up that is not working. Maybe the bridge you had to solder to use the HAT PWM setting is not working properly.
You need to systematically rule out the mistakes.

hzeller, 2016-10-13:
Haven't heard anything, I presume you figured it out.

closed 2016-10-13 by hzeller

dennys78, 2016-10-13:
Great the bridge not working properly!!! Now all ok. Thank for the support!!!!!