hzeller/rpi-rgb-led-matrix #232: LED Matrix and 1wire temperature sensor

viper2035, 2016-10-13:
Hi,
first thank you for all the work you put into the project. I love it :-) I use my led matrix to show a screen with time, outside temperature (value from a weather site) and the inside temperature (value from a 1wire temperature sensor DS1822). The problem is, that there is some kind of blocking the 1wire system if the matrix shows a picture. Normally the temperature value gets written into this file "/sys/bus/w1/devices/"SENSOR-ID"/w1_slave" But if the led matrix shows something the value gets wrong...something like 1048 °C or -10°C. If I stop the LED Matrix the temperature gets back to the right value. Do you know a solution for that problem?

hzeller, 2016-10-13:
Where is the sensor connected ? Do you happen to use a GPIO that is also used by the LED matrix ?

hzeller, 2017-07-30:
So turns out that 1 wire connection is messing with the LED matrix, so you need to somehow differently connect the wires.

closed 2017-07-30 by hzeller