hzeller/rpi-rgb-led-matrix #234: Problems with colors and flicking

nullday000, 2016-10-14:
Hey there!
I use led-image-viewer to display some pics on my screen and I faced some problems.

I have my LED panels connected to raspberry3 using this adapter:
http://imgur.com/afW2X2Z

1. Color issue. I tried to display a pic with rgb matrix and figured out that my colors are messed.
The original pic is here:
http://imgur.com/t7dVFwl

So I launched it like this:
`sudo /home/pi/ledmatrix/rpi-rgb-led-matrix/utils/led-image-viewer /home/pi/rgb.png --led-rows=32 --led-parallel=3 --led-chain=8 --led-brightness=60`
And here is the result:
http://imgur.com/sNLUatO

After that I launched it like this:
`sudo /home/pi/ledmatrix/rpi-rgb-led-matrix/utils/led-image-viewer /home/pi/rgb.png --led-rows=32 --led-parallel=3 --led-chain=8 --led-brightness=60 --led-swap-green-blue`
and the result is here:
http://imgur.com/UvOdHNp

So, as you can see none of these tries gave me right results.. So I have no idea how to fix it.

2. Flickering. Any pic I send to the screen flickers every 3-4 sec. I tried to use different options such as brightness, led-pwm-lsb-nanoseconds, led-no-hardware-pulse, led-slowdown-gpio. Nothing helps. If I use led-show-refresh it shows 143Hz. If I use options that I mentioned above it can be from 100 to 230Hz but it doesn't solve flickering problem. 
The only thing I noticed is that if I use --led-pwm-bits=2 it shows 315Hz and flickering disappears. If I use --led-pwm-bits=4 it shows 243Hz and I can notice flickering again.

Thanks you guys in advance. Will appreciate any help and any ideas.

hzeller, 2016-10-14:
* I don't know that adapter, where did you get it from ? (The 'official' ones [are in the project](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter), but this doesn't look like it).

 * Interesting, looks like that adapter or your particular LED panel is messing up the sequence of the color compared to the usual. There is no direct flag to change that (this is the first time I hear about this swap), but you can work around that by modifying `lib/hardware-mapping.c` : https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c#L44  There, you need to swap around the GPIO values for red and blue, and then give it the `--led-swap-green-blue` flag to do the rest (or, alternatively, you can swap all around in one swoop and don't use the swap green/blue option).

So this
```
    .p0_r1         = GPIO_BIT(11),
    .p0_g1         = GPIO_BIT(27),
    .p0_b1         = GPIO_BIT(7),
    .p0_r2         = GPIO_BIT(8),
    .p0_g2         = GPIO_BIT(9),
    .p0_b2         = GPIO_BIT(10),

```
becomes this:
```
    .p0_r1         = GPIO_BIT(7),  // <- this was before in blue
    .p0_g1         = GPIO_BIT(27),
    .p0_b1         = GPIO_BIT(11),  // <- this was before in red
    .p0_r2         = GPIO_BIT(10),   // <- this was before in blue
    .p0_g2         = GPIO_BIT(9),
    .p0_b2         = GPIO_BIT(8),  // <- this was before in red

    /* also below with the other red and green in p1 and p2 */
```

  * Flickering: Check what CPU usage you see. If it is solid 100% for the core displaying, then the kernel sometimes decides to annoyingly interrupt it (you can see the core view in `top` by toggling with the key `1`). A little help but probably not fully would be to use the `isolcpus` setting mentioned in the README.md. You can see what each CPU thread is using with uppercase `H` in top. Usually, this should not really happen unless you have many displays chained.

  * For a Raspberry Pi 3 you might need `--led-slowdown-gpio=2`

  * Also, if you have not disabled your sound system, you might see similar interruptions (see Troubleshooting area in the README). You must make sure that no `snd_bcm2835` is loaded, otherwise things won't work properly.

nullday000, 2016-10-15:
First of all, thank you for such an instant reply. I appreciate it.

I got this adapter from my Chinese partners. Have no idea how they built it.
Btw I don't know considers my display big or not, but it has about 32 led panels.

Tried the fix you suggested, after that I called make clean in root folder and after that make, it compiled everything, after that I went to utils folder and called the same, got new binary and tried to launch with the same options - nothing changed.

Top shows that led-image-viewer uses approx 70-75% of cpu... So I followed your tip and made isocplus and rebooted the raspberry and tried again. And top as I can see in top only cpu3 is busy up to 75%, so it worked I guess. But nothing changed actually, it flickers as it was before.. :(
And I can see no spikes in top, so led-image-viewer is the most cpu using process, everything else is less than 1%.

Tried --led-slowdown-gpio=2 as well, no changes with flickering.

Sound system was disabled completely like you described in README.

hzeller, 2016-10-15:
If you have a link to the board or can provide me one that would be good, so that I can put it on my todo list to make things work with that.

Did you call 'make' in the utils/ folder ? Just typing make in toplevel compiles the library.

With `--led-parallel=3 --led-chain=8 --led-rows=32 --led-slowdown-gpio=2` you should get around 180Hz refresh rate and about 60% CPU. With slowdown-gpio=1 it is almost 200Hz refresh rate and a little less CPU.

Another thing what I have seen people run into is if they overclock the Pi. That throws off internal timings so does all kinds of crazy things. Don't overclock.

nullday000, 2016-10-15:
My partners say that their board is 100% like yours.
I just made a picture of this board, you can find it here:
http://imgur.com/kAyC9zr
http://imgur.com/EMPCQ3I

Yeah, I've called make everywhere.
Also, you wrote:
>  /* also below with the other red and green in p1 and p2 */
so, I got next code, is it right? 
Im not sure because you changed red and blue but after that you write about red and green 😕 
`    /* Parallel chain 0, RGB for both sub-panels */
    .p0_r1         = GPIO_BIT(7),  /* masks: SPI0_SCKL  */
    .p0_g1         = GPIO_BIT(27),  /* Not on RPi1, Rev1 */
    .p0_b1         = GPIO_BIT(11),   /* masks: SPI0_CE1   */
    .p0_r2         = GPIO_BIT(10),   /* masks: SPI0_CE0   */
    .p0_g2         = GPIO_BIT(9),   /* masks: SPI0_MISO  */
    .p0_b2         = GPIO_BIT(8),  /* masks: SPI0_MOSI  */

    /* All the following are only available with 40 GPIP pins, on A+/B+/Pi2,3 */
    /* Chain 1 */
    .p1_r1         = GPIO_BIT(6),
    .p1_g1         = GPIO_BIT(5),
    .p1_b1         = GPIO_BIT(12),
    .p1_r2         = GPIO_BIT(20),
    .p1_g2         = GPIO_BIT(13),
    .p1_b2         = GPIO_BIT(19),

    /* Chain 2 */
    .p2_r1         = GPIO_BIT(3), /* masks TxD when parallel=3 */
    .p2_g1         = GPIO_BIT(2),  /* masks SCL when parallel=3 */
    .p2_b1         = GPIO_BIT(14),  /* masks SDA when parallel=3 */
    .p2_r2         = GPIO_BIT(21),
    .p2_g2         = GPIO_BIT(16),
    .p2_b2         = GPIO_BIT(26),`

`$ sudo /home/pi/ledmatrix/rpi-rgb-led-matrix/utils/led-image-viewer /home/rgb.png --led-rows=32 --led-parallel=3 --led-chain=8 --led-swap-green-blue --led-show-refresh --led-slowdown-gpio=2
Size: 256x96. Hardware gpio mapping: regular
Load images...
Display.
 116.1Hz`

I didn't overclock it.. Just a usual raspberry with raspbian.

hzeller, 2016-10-15:
that board has six connectors, while the other boards are just 3 connectors - are these just mirrored in the board you have ?

On a Raspberry Pi 3, I get about 170Hz with the settings you mention above, not 116Hz so something is strange, as if it has too slow memory or something. I usually suggest raspbian lite, but I don't think that explains the difference in speed.

nullday000, 2016-10-15:
yeah, I have two standalone panels and they are mirrored, that's probably why it's 6 connectors..

so, do you have any ideas about the problems?

hzeller, 2016-10-15:
You need to find out why your system is so slow. It is strange that it is only 116Hz compared to a typical Raspberry Pi3. Make sure to disable all things inside the Pi that could interfere (for instance, don't install raspbian but raspbian lite). Don't run a graphical user interface etc.

hzeller, 2016-10-15:
Your code changes look good, so that should give you the right colors after recompilation.

nullday000, 2016-10-16:
Just fixed my problem with colors, in your original code I had to change red with green and it helped me.

Talking about flickering - I got you. Will try to work on it.
Thank you for your support. Your work is amazing.

closed 2016-10-16 by nullday000

ArsalanDotMe, 2016-12-29:
@nullday000 I ordered some LED panels and an adapter from China and they sent me the same adapter as you have! Probably not co-incidently, I have the same flicker problem as you have. Maybe its due to the adapter?