hzeller/rpi-rgb-led-matrix #235: Update Only Part of Screen Python

sepehr500, 2016-10-15:
I am trying to make a simple clock using the python wrapper, but I can not get only part of the screen to update without the rest flickering. At first I tried using the PIL image support, but calling `matrix.Clear()`clears the whole screen causing a flicker when redrawing. Next I tried the canvas wrapper, and that does let you add things to the screen without flicker, but I did not see a way to remove an object after it has been drawn, or a way to move the object after it has been drawn. My goal is really to have a clock at the top of the sceen, and a scrolling message at the bottom, without the clock at the top flickering. I do not think the flicker is because of my pi's performance. Any idea how to implement this?

hzeller, 2016-10-15:
I would do the double buffering, and then draw each frame with timestamp and the additional content below, then swap images. That way, you never have flicker and have a smooth animation (since you are drawing on an off-screen, you can call Clear() and redraw, taking your time, as the other frame is still displayed). Also, you can have easily the clock in foreground transparently hovering over the other content because you re-write it everytime (otherwise it would not really be possible to transparently 'erase' a previous time).

If you don't want to do that, then you should just clear the area in the corner with the text with black or whatever background color you like. The c++ API has a convenient DrawText() call that allows you to set the background color for that. The Python API does seem to be still on some older version of the API, so in that case you have to manually black out the area.

sepehr500, 2016-10-15:
Worked out great! Thank you so much!

closed 2016-10-18 by hzeller