hzeller/rpi-rgb-led-matrix #237: Pixels 17-32 always blue

gregfriedland, 2016-10-18:
First off, thanks for providing this library to the community; it has been an excellent resource for me to get up and running quickly with 9 panels at once.

My setup is an RPi 3 running rasbpian lite with 3 parallel lines of 3 64x32 panels each with a 40A power supply. I'm using your active adapter with 425HCTs and 4600uF caps on each of the panel power inputs.

I'm posting this issue to see if you have any suggestions to get rid of a persistent blue band across the bottom half of the top 3 panels of my 3x3 64x32 panel array. All the pixels in rows 17-32 appear to be lit up all the time. Well not all the time to be exact, this behavior doesn't happen when I first start using the board, but as it "warms up", these pixels begin to flicker blue and then after a few minutes, they are always blue. All the other pixels appear to be fine, and even the red and green channels on those pixels seem ok. I checked the soldering of the P0_BX pins on U1 but don't see any obvious bridges or shorts. Any suggestions on other things to try or look for? If all else fails, I will resolder a new active adapter but surface mount soldering is not my forte so I'd like to avoid that if possible.

Thanks.

hzeller, 2016-10-18:
So sounds like somewhere on the path between the Pi and the first matrix something is wrong on the **`P0_B2`** line - if it is all lit, it sounds like it is just an open connection which then is interpreted by the input as 'high'.

First you want to circle in if this is due to the panel: swap the chains and connect another chain that usually is working to the first chain. If that also turns blue, then it is something on the adapter. If it works, then this hints that the first panel in that chain might be broken. You can confirm if  you connect it to some other chain and see if it fails there as well.

So if

 * The panel is broken: have a closer look at its input connector and chips. Maybe it has some loose soldering joint at its input chips ? Or a loose connector ? The panels typically have surface mount IDC connectors, maybe one of the pad solders has a miniature fracture which is hard to see, but becomes a problem when it warms up. Some manual touch-up of these solder joints might be good.
 * If the signal out of the adapter is not good, then double checking the soldering joints and also Pin 19 on the GPIO connector is probably good. Looks like you already did that.

Maybe, these panels are simply too slow (the Raspberry Pi3 is often a little too fast for many panels), so you could try `--led-slowdown-gpio=2`.

Check the voltages: if the wires are too thin, you might not get full 5V at the panels. Measure the voltage after you show a full white picture.
Slighlty increasing the voltage (say 5.25V) at the power supply might help (but don't turn it up too much: only when a lot of LEDs are on, the voltage drops, otherwise it will go up to the voltage you set at the power supply). Lower voltages make the logic chips slower, so if the voltage is too low, that could also be a problem if it is due to speed.

Another thing to double check: Make sure that all panels have the ground connected together and that properly connected to the GND of the adapter (high current situations can make the GND voltage level be different in different places, which can make interpreting logic signals challenging).

Also good for better ground connection: if you haven't already, make sure to connect the pin 4 and 8 to GND in the [E-config jumpers](
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3#choose-e-line-for-64x64-panels-with-132-multiplexing) - first picture.

Overall, I suspect the most likely reason is a hair fracture on Pin 7 of the IDC connector on the matrix.

gregfriedland, 2016-10-19:
Thanks for the detailed response! I will try your suggestions tonight.

gregfriedland, 2016-10-21:
I tried your debugging suggestions and actually the panels seem ok. The blue pixels moved with the adapter board output when I rewired things. I looked closely at all the solder joints on the board under magnification and didn't see anything off, but I went ahead and resoldered all the P0_B2 ones anyway on the board and connectors. And the blue pixels appear to be gone!

I was close to considering it completely working except now after a longer warmup time than before (~30mins), all rows in the first chain are randomly changing color. It seems like all pixels in a row are the same color at any given time, but that color changes randomly from row to row and time to time.

I'll try resoldering the connections for P0 that I haven't yet and also check the voltage drops at high brightness as you suggested.

gregfriedland, 2017-03-07:
I resoldered things and that didn't fix the problem until I tried plugging into a different power outlet, which fixed all the issues I was experiencing. Strange and inexplicable but at least it's working now. :)

closed 2017-03-07 by gregfriedland