hzeller/rpi-rgb-led-matrix #238: 15 panels 5x3 setup 160x96 pixels can't rotate (Adafruit HAT)

mattyams, 2016-10-23:
Thank you for the great code it is exactly what I am needing, I hope. As the title says I have 15 panels chained together and in a 5x3 configuration. I am getting images but the are not aligned properly. I have read about eh canvas transform but I just can not figure out how to do it.  I would get the adaptor you have posted but I honestly do not have the time. Stupid deadlines. Any assistance would be super helpful!

Thanks!

hzeller, 2016-10-23:
The canvas transform essentially takes an input coordinate and says where the output coordinate would be.
In your case, you would probably have X = X % (5*32) and Y = X / (5*32) + Y
If you have a zig-zag configuration (so the second row goes backwards), you need to mirror that one.

mattyams, 2016-10-23:
Yeah, I see what you are saying but honestly....I don't get it. Hah.  I have hobbiest coding knowledge but I am no pro by any means.

hzeller, 2016-10-23:
Read the other canvas transformer implementations. This is pretty basic stuff which you'll figure out quickly.

mattyams, 2016-10-23:
Yeah....I have no idea.

mattyams, 2016-10-31:
Ordered the active adapter from OSH. Gunna give it a go.

closed 2017-07-30 by hzeller