hzeller/rpi-rgb-led-matrix #239: How to correctly shutdown LED matrix code

alexlukelevy, 2016-10-24:
First of all, thanks for such an amazing library!

I've had some success using the python bindings to write to a 32x16 matrix with my Raspberry Pi B+ however I seem to have a problem after I've killed the process. 

After killing the python process, the matrix clears (as expected) but then several hours later, ghost text starts appearing on the matrix (even when the matrix has no power). I've also noticed that I am now unable to SSH to my Raspberry Pi. 

Is there a buffer I need to clear on process exit? Or could this be a memory leak you've witnessed before?

hzeller, 2016-10-24:
If you see actual _text_, then the matrix is still running and you need to check your shutdown procedure. You should not just kill the process, but exit regularly (it internally does the clearing and shutting down a thread). In `top`, you can see if there is still some CPU used (because the display update takes some CPU).

If you just see ghosting LEDs show up, then you probably have simply a floating Output-Enable line. It needs to be pulled up (this is the 10k resistor in the active adapter board). Also, if you have other processes running that make use of the GPIO pins after you stop the program, then they can of course also influence an attached Matrix.

(if you don't power the matrix, then even the faint digital signals from the active adapter board pump power into the chips, strong enough to have the LEDs light up faintly. If you have the matrix directly connected to the Pi, then the power comes from the pi and you should avoid this as much as possible as the output drivers are directly on the SoC which you don't really want to sacrify).

alexlukelevy, 2016-10-25:
Thanks for your response. 

What you described in your second paragraph sounds like it might be worth trying as I'm seeing ghosting. Unfortunately I'm a newbie with electronics, so I'll have to look up what you meant by pulling up the line, but it sounds promising.

hzeller, 2016-10-25:
Essentially it means that you connect the output enable wire not only to the LED panel but also via a resistor to a logic high signal. So a 10k (=kiloohm) resistor (does not need to be exact; something between 4.7k and 15k will probably work) between output enable (pin 12 on the GPIO header) and 3.3V (pin 1).

What does it do ? When the output is not driven by the Raspberry Pi, the resistor makes sure that the Panel always 'sees' a high signal. Because a low signal tells the panel to switch on the LEDs.

It is called 'pull-up', because it pulls the voltage towards the high signal.

alexlukelevy, 2016-10-25:
Awesome, thanks for the explanation! I bought a 10k resistor today, so fingers crossed that does the trick.

closed 2016-10-25 by alexlukelevy