hzeller/rpi-rgb-led-matrix #240: Adafruit HAT

crablab, 2016-11-01:
Hi, 

The updated library now no longer works with the Adafruit HAT - https://goo.gl/zlhi9a 
Is there a list of wiring changes that have been made so I can try and work out what I need to do, to get this working again? 

Thanks,

depili, 2016-11-01:
Reading the readme might help... there is a whole section in it about the adafruit hat.

crablab, 2016-11-01:
I have - I wanted to know why the wiring has been changed and what exactly has changed.

depili, 2016-11-01:
Then look into wiring.md

crablab, 2016-11-01:
Yeah sure, thanks for your help - I have already looked around. I'm interested in understanding exactly what changes have been made and what these changes have accomplished. (There is no changelog)

hzeller, 2016-11-01:
You are not saying what versions you are comparing and which version you used before that was working for you.

In this library, since the third-party Adafruit HAT is supported, you edit the `lib/Makefile` and change that to support that board. That never changed; so whatever you did before should work now.

In newer versions (a couple of months), you can _also_ choose that on the command line if you are using C++ with `--led-gpio-mapping=adafruit-hat` or `--led-gpio-mapping=adafruit-hat-pwm` depending if you were using the modification, which makes that even simpler.

The default wiring of this library is what is described in wiring.md, Adafruit used their own wiring, so it is always needed to change switch to that third-party wiring.

(Note, from the link you provided it looks like you have read the Adafruit HAT documentation, then this is _never_ correct if you are using this original library. This project here is the original libray. The library in the Adafruit documentation is a fork from a very old version of this library which they then hardcoded to work with the HAT - so it works right out of the box if you are using that fork)

closed 2017-07-30 by hzeller