hzeller/rpi-rgb-led-matrix #242: problem with P10 SMD scan1/4 OUTDOOR

blaster22, 2016-11-09:
Hello,
i bought a P10 SMD 32x16 (LED 3535) scan 1/4,  I have Rpi-3 Model B andHUB75 from adafruit. 
I have problem with using library. 
I read from: https://github.com/hzeller/rpi-rgb-led-matrix/issues/136 and ask to Mr. Zeller that for use my panel i can edit CanvasTransformer.
But i don't undestand how i can edit it. I'm not very practical to write code and compile it.
Can someone help me?

In attachment there is a table with one list of led, the first led is number 1, second number 2 etc..

in attachment there are video and photo of my panel.
https://www.youtube.com/watch?v=eNP6_jn1rC0
[image]
[image]
[image]
[image]

thanks

fulviolesca, 2017-01-11:
Hi
I have the same problem with P10 panel from China. Did you solve ?

blaster22, 2017-01-11:
Hello,
i don't have solved.
are you italian?

fulviolesca, 2017-01-11:
Hi
yes. I'm italian...

VolkerGoeschl, 2017-02-06:
Hi Guys
I think I have panels with the same arrangement and I have already written a transformer for them.
I will post the code later today.
Cheers
Volker

fulviolesca, 2017-02-06:
GREAT 
This would be really great !!

VolkerGoeschl, 2017-02-06:
Here you go:
add this to lib/transformer:
```
/********************************/
/* P10outdoor Transformer Canvas */
/********************************/
class P10outdoorTransformer::TransformCanvas : public Canvas {
public:
  TransformCanvas() : delegatee_(NULL) {}

  void SetDelegatee(Canvas* delegatee);

  virtual void Clear();
  virtual void Fill(uint8_t red, uint8_t green, uint8_t blue);
  virtual int width() const;
  virtual int height() const;
  virtual void SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue);

private:
  Canvas *delegatee_;
};

void P10outdoorTransformer::TransformCanvas::SetDelegatee(Canvas* delegatee) {
  delegatee_ = delegatee;
}

void P10outdoorTransformer::TransformCanvas::Clear() {
  delegatee_->Clear();
}

void P10outdoorTransformer::TransformCanvas::Fill(uint8_t red, uint8_t green, uint8_t blue) {
  delegatee_->Fill(red, green, blue);
}

int P10outdoorTransformer::TransformCanvas::width() const {
  return delegatee_->width() / 4;
}

int P10outdoorTransformer::TransformCanvas::height() const {
  return delegatee_->height() * 2;
}

void P10outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  int new_x = x;
  int new_y = y;
  int panel_x = x / 32;
  int x_offset = 0;
  switch (y % 8){
    case 0: x_offset = 64;break;
    case 1: x_offset = 64;break;
    case 2: x_offset = 72;break;
    case 3: x_offset = 72;break;
    case 4: x_offset = 0;break;
    case 5: x_offset = 0;break;
    case 6: x_offset = 8;break;
    case 7: x_offset = 8;break;
  }

  new_y = 4 * (y / 8) + (y % 2);

  if ((y % 4) < 2) {  //countdown
    new_x = ((x % 32) * -1) + (24 * (((x % 32) / 8) + 1)) - 17;
  } else {
    new_x = (x % 8) + (16 * ((x % 32) / 8));
  }
  new_x += x_offset;
  new_x += (panel_x * 128);
  delegatee_->SetPixel(new_x, new_y, red, green, blue);
}
/*************************/
/* P10outdoor Transformer */
/*************************/
P10outdoorTransformer::P10outdoorTransformer()
  : canvas_(new TransformCanvas()) {
}

P10outdoorTransformer::~P10outdoorTransformer() {
  delete canvas_;
}

Canvas *P10outdoorTransformer::Transform(Canvas *output) {
  assert(output != NULL);

  canvas_->SetDelegatee(output);
  return canvas_;
}
```
and this to /include/transformer.h

```

class P10outdoorTransformer : public CanvasTransformer {
public:
  P10outdoorTransformer();
  virtual ~P10outdoorTransformer();

  virtual Canvas *Transform(Canvas *output);

private:
  class TransformCanvas;

  TransformCanvas *const canvas_;
};
```

...then add the following to use the transformer:

```
rgb_matrix::CanvasTransformer *transformer = new rgb_matrix::P10outdoorTransformer();
matrix->SetTransformer(transformer);
```

Be aware that you must set rows to 8 and the chain length to 4 times the number of panels you are using

I´m currently setting up a 5 by 2 -Panel Display using this transformer AND the UArrangement-Transformer.

VolkerGoeschl, 2017-02-06:
@hzeller : would you be so nice to check my SetPixel code for obvious performance-optimizations?
(the last time i have touched C was at the university 20 years ago ;-) )

hzeller, 2017-02-06:
Don't worry about speed @VolkerGoeschl , the implementation only calls it once in the beginning to internally build up a look-up table.

BTW, you should call `ApplyStaticTransformer()` instead of `SetTransformer()`, which used to be how that function was called a while back, so SetTransformer() is now deprecated.

In that case, you don't even need to `new` the object, but can do
```
  matrix->ApplyStaticTransformer(rgb_matrix::P10outdoorTransformer());
```

https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates

fulviolesca, 2017-02-07:
WOW !!!!!!
All is working good now !!!!
Tanks to VolkerGoeschl and expecially to Hzeller 👍 
I'll need some additional help 'cause I have 5 row every of 5 P10 display, I read what I need to do but some code woud ne appreciated :)
Thanks a lot

hzeller, 2017-02-07:
Code is similar to the other transformer: you have to map pixels from where you get them to where they should be.

You could connect them in three parallel chains with two each in a chain (well, the last one only has one), then use the UArrangementTransformer:
```
   matrix->ApplyStaticTransformer(rgb_matrix::P10outdoorTransformer());  // fix outdoor pixels
   matrix->ApplyStaticTransformer(UArrangementTransformer(3));              // U-arrangement for three
```

fulviolesca, 2017-02-07:
Hi 
I have 5 row each  of these with 5 panel , 25 total., as below

[05]<[04]<[03]<[02]<[01]<RASP 1
[10]<[09]<[08]<[07]<[06]<RASP 2
[15]<[14]<[13]<[12]<[11]<RASP 1
[20]<[19]<[18]<[17]<[16] not connected
[25]<[24]<[23]<[22]<[21] not connected

How must connect them ?  I suppose a chain of 8, as the second and the last with 9 , but which way?
05 has to go to 10>09>08 ? And the other ???

Unfortunatly I'm a real newbie with C (i use Delphi) and circuit so I need to ore all you with my questions .. SORRY :)

hzeller, 2017-02-07:
You could of course do a 2x8 and 1x9 arrangement which is the best for refresh rate, but then have to write your own code. If you use the template of how the other transformers are written, you can also totally do that with your Delphi knowledge - the basic operations are the same. So no excuses :)

If you don't want to write code, then use the UArrangmentTransformer. You do that with three parallel chains of 10. Two of them with actually 10 connected and one chain where only 5 are connected. Then apply the UArrangementTransformer(3). It will then show as 6 panels high (because it still thinks the last has 10 panels), but that is the price of using the built-in solution.

closed 2017-02-07 by hzeller

gausie, 2017-02-19:
Is there any chance of having this "snake"-style transformer tidied up and shipped with the core library?

getvladimir4, 2017-11-14:
Sorry,
I want to use the VolkerGoeschl solution but I do not know how to execute it, they could help me especially the last part ...

in what file should I add this?
[image]

getvladimir4, 2017-11-14:
This solution I think can help me. But I do not know how to proceed and try it on the led panel that I have. 

This image results when executing
sudo ./demo -r 16 -c 1 -D 3
[image]

and that...
sudo ./demo -r 16 -c 1 -D 1 runtext16.ppm
https://www.youtube.com/watch?v=zr0klS0OVoE&feature=youtu.be

* My led panel is P8 HUB75

I see that @fulviolesca  worked on @VolkerGoeschl solution but I would like to know how he proceeded?
But in the end @hzeller  suggests something and that part I do not understand ...

Thank you very much for the comments you can give me. It will be very useful.
Vladimir,

Arsenik4444, 2017-12-12:
Hi Sir, i did read all issues worked by others in the topic and the items you have send to me, thanks a lot, then i have see a transformer which is made and published by VolkerGoeschl and which may work for the panel i'm using but going through all files i do have few questions regarding the steps:
-what have to be done replacing or just adding the files in the corresponding location:
*where to add this to lib/transformer.cc: in the code
*where to add this to /include/transformer.h: in the code
*where to add this location and area:
rgb_matrix::CanvasTransformer *transformer = new rgb_matrix::P10outdoorTransformer();
matrix->SetTransformer(transformer);
*and after all done, what will be the final steps to activate the modification
thank you Sir.

dl2ocb, 2018-02-23:
Hello,

i got some of this nice P10 Outdoor Panels too. 

They are 1/4 Scan Panels and i inserted the Code for the P10 Transformation provided by VolkerGoeschl (many many Thanks). 

My HW-Config is a Raspberry Pi 3 with the active Adapter for 3 Chains and 1 Panel connected for testing.

I tested the demo Application after inserting the following Line into it:

... Upper Code ....
if (rotation > 0) {
    matrix->ApplyStaticTransformer(RotateTransformer(rotation));
  }

// INSERTED by DL2OCB --------------------------------------------------
matrix->ApplyStaticTransformer(P10outdoorTransformer());
//------------------------------------------------------------------------------------

  printf("Size: %dx%d. Hardware gpio mapping: %s\n",
         matrix->width(), matrix->height(), matrix_options.hardware_mapping);
... more Code ....

If i got it right, this should transform the Matrix to be displayed the rigth way on an 1/4 Scan P10 Panel.

I used this commandline:

sudo ./demo -D3  --led-rows=8 --led-chain=4 

The Output of the Panel looked like this:

[image]

I dont know what went wrong ... @hzeller can you give me a hint please ?

I can provide you a Panel for testing if you need.

Many thanks for the nice Lib !!