hzeller/rpi-rgb-led-matrix #243: Detach TTY and update PPM on demand - Feature request

RamyashreeGR, 2016-11-17:
Hi,

We wanted the demo executable to accept ppm on demand i.e., Without receiving an interrupt from keyboard, the current scrolling text should change to the new PPM when executed, while the same demo process is running in background should be overridden[Detached from TTY].

Is there a way to figure this out?

Thanks,
Ramya.

hzeller, 2016-11-17:
Just write a little server that receives a new image from the network, e.g. via UDP.

depili, 2016-11-17:
See https://github.com/depili/rpi-matrix for example of a really simple udp server.

RamyashreeGR, 2016-11-18:
@depili Seems interesting. Can you explain how to use your source or probably a quick start should suffice.

Thanks,
Ramya.

gadag99, 2016-11-23:
Hi , I tried execute with your provided link , but I am not able to get output on Led-display , So could you please suggest me how to send command to upd file ? also can i send ppm to udp sever?

Gadag007, 2016-11-23:
Hi , I tried execute with your provided link , but I am not able to get output on Led-display , So could you please suggest me how to send command to upd file ? also can i send ppm to udp sever? if u tell one eg that could be more helpful  for me.

Thanks and regards 
Praveen

depili, 2016-11-23:
@RamyashreeGR clone, pull the submodule, make, run, send the data as udp packets to port 9999 as 8bits per color, 24bits total per pixel. it's just a minimal simple udp listener.

I myself have written the clients itself in go

Gadag007, 2016-11-23:
@depili does your program works for ppm image file?

Gadag007, 2016-11-23:
@depili , i cloned , pulled the sub modules, make and run is performed , I got stuck with the format of sending data to 9999 port for your program, I will be glad if you provide me the command line parameters how you sent , this one command line example format will solve my problem.

Thank you
Praveen

RamyashreeGR, 2016-12-02:
@hzeller  @depili  
Hi,
Thanks for the valuable hints and support, now my server program works for sending ppm files also.

Thanks
Ramya

hzeller, 2016-12-02:
Cool! Looks like we can close this now then :)

(PPM support would essentially be the FlaschenTaschen protocol that I implemented in a different project).

closed 2016-12-02 by hzeller