hzeller/rpi-rgb-led-matrix #244: Active-3 board: Use of spare gpio pin for E line

gpulido, 2016-11-18:
Hello,
First of all, thank you for this amazing library and side projects boards.
I have just bought 32/16 p10 panels 1/4 scan to be used with raspi 2 / 3 for a timer, but I also need to control an on/off relay, so I would need just one GPIO free.
I have take a look to yhe Active-3 board, and since the panels I've bought doesn't need a E-line, Is there any way to use such pin (pin10 / GPIO15) to wire a Relay?. If so, can I just solder a connector to the E pin instead of shortcut it to ground?
Again thank you very much for the support
Gabriel

hzeller, 2016-11-18:
Yes. The buffered GPIO15 is given at the free jumper pins, so you have a conveniently way to solder to it. You still would jumper the 4/8 to GND (these are passed to the header).

The library will not configure the GPIO15 if not needed, so you are free to use it independently.

gpulido, 2016-11-21:
Thank you for the quick answer.
I already have noticed the RxD jumper placeholder, so just to be sure:
I have to jump the 4 and the 8 pin to GND (as in the picture "No 1:32 64x64: to GND"
And where I should wire the control pin? I have two "E" placeholders, just to one of them? or I have to solder to the gpio pin directly?

closed 2017-07-30 by hzeller

gpulido, 2019-05-22:
Hello again,
My project is advanced, and I just noticed that you have added some code to control the free gpio's without the need o using another library (wirinpi for example).
I need to use the GPIO15 as output to control a relay, but I haven't found any examples of how to do it with your library :(
The whole picture, I already have a udp multicast listening for RGBRGB... data, and I need another multicast socket port listening of data that would trigger the GPIO15. I can do it using python and the pigpio library but as the matrix library can provide it, it would reduce the amount of code to compile / install / deploy.