hzeller/rpi-rgb-led-matrix #245: 2x 64x32 panels flipped

GumboBeans, 2016-11-24:
Hi, 

I have 2 64x32 panels, which seems to be working, however, the image is not shown correctly, i'm sure i am missing an cli argument somewhere, but i cant figure it out, hopefully someone here can help me. 
Its looks to me like the "U" arrangement of the panels is the wrong way around, but i have the boards wired up correctly from what i can see. 

Board: Raspberry Pi Zero
Command: sudo ./demo -L --led-rows=32 --led-chain=4 -D0

Many thanks for a great project :)

Picture below is the D0 example. 
[image]

hzeller, 2016-11-24:
Why don't you just fold the panels the other way around ?

GumboBeans, 2016-11-25:
Yeah... i now feel like an idiot. Sorry for wasting your time, Flipping the panels worked.

closed 2017-07-30 by hzeller