hzeller/rpi-rgb-led-matrix #246: No pwm_lsb_nanoseconds in Python

xboomx, 2016-11-24:
Am I missing something? I couldn't find a way to control pwm_lsb_nanoseconds parameter with Python. Is there any way of reducing ghosting when using Python?

hzeller, 2016-11-25:
Not all flag configuration options are available in Python (like the pwm nanoseconds), but I am happy to apply a pull request that provides the options struct in the Python wrapper.

For now, you have to change the setting statically in lib/Makefile and recompile the library and Python wrapper.

xboomx, 2016-11-25:
Thank you! You're the best :)

hzeller, 2017-01-25:
In the latest commit https://github.com/hzeller/rpi-rgb-led-matrix/commit/e2ffd91644eb9ca36b57921c07106be1ef0d0f63 there are now a couple of options available in Python, including `--led-pwm-lsb-nanoseconds`

closed 2017-01-25 by hzeller