hzeller/rpi-rgb-led-matrix #249: (Python) Getting unrecognized argument --no-led-hardware-pulse

Gariantroll, 2016-12-01:
I am getting this error.  I tried to blacklist the sound, but I am still getting this error

runtext.py: error: unrecognized arguments: --led-no-hardware-pulse

hzeller, 2016-12-01:
Yes, this is expected, the Python library does not use the flags yet. Pull requests that change that welcome.

Gariantroll, 2016-12-01:
I got rid of the audio file and it worked.  Thanks for your help and quick responses.

hzeller, 2017-01-25:
Getting rid of the audio is the best solution anyway.

In the latest commit https://github.com/hzeller/rpi-rgb-led-matrix/commit/e2ffd91644eb9ca36b57921c07106be1ef0d0f63 the lack of options has been fixed, and there is now a `--led-no-hardware-pulse`

closed 2017-01-25 by hzeller

tomsepe, 2023-11-14:
Hello, It seems this error is back. I have a raspberrypi 4 and just downloaded the files using the script provided by adafruit's tutorial. and when I try to run runtext.py with the "--led-no-hardware-pulse" I get the "expected one argument response"