hzeller/rpi-rgb-led-matrix #250: (Python) Background Manipulation

Gariantroll, 2016-12-01:
I am trying to find out where I can manipulate the color of the background when the matrix running.  I would like to set it to off or to black.

hzeller, 2016-12-01:
You have control of the color of every pixel; SetPixel() will set the color. There is no additional thing needed to distinguish background/foreground - if you don't set a pixel at all, it is simply black.

(If you have inverse data lines, the display will show inverse colors, e.g. a white background, in which case you'd choose the 
`-DINVERSE_RGB_DISPLAY_COLORS` compile option.)

Gariantroll, 2016-12-01:
Bro, you are superman of the matrix world.

Gariantroll, 2016-12-01:
Quick Question, I am trying to find a way to use the textrun.py sample and make the background LEDs one color but the text another color.  Can you point me in the right direction on this?  I am guessing that after the Text prints I have to make every other background pixel soemthing else, but I am wondering how to approach this.

hzeller, 2016-12-01:
The C++ library has a function for that:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/graphics.h#L76

The simplified Python interface has not, but it should be pretty straight forward for you to also wrap this function if you use Python.

Gariantroll, 2016-12-13:
Hey hzeller,

I am having an issue as I came back to this project.  I was trying to write the wrapper then I realised  that the I must have something wrong since the default background color is supposed to be black, but instead it is beaming white.  I think I have a data line issue as you have implied in a previous response, so since I tried to implement that compile option, but it is not recognising the -DINVERSE_RGB_DISPLAY_COLOR.  Where am I supposed to use this option?

hzeller, 2016-12-13:
You have to uncomment
  https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L39
then recompile the library and the Python wrapper.

Gariantroll, 2016-12-13:
Just to be clear, is the Python Wrapper where I use 
make build-python PYTHON=$(which python3)
sudo make install-python PYTHON=$(which python3)

or am I just typing "make" in the python directory

hzeller, 2016-12-13:
yes. Essentially change the makefile, type make clean in the lib directory and then go through the steps needed for building the Python lib again.

Gariantroll, 2016-12-13:
I have tried  -DINVERSE_RGB_DISPLAY_COLOR commented out and active, but I am still getting the bright light background.  I know I am missing something, but I am stumped for now.  Any suggestions?

hzeller, 2016-12-13:
Systematically rule out possible problems. Start with running the c++ demo program to verify that your display now shows the correct colors. Also verify with the led-image-viewer. Read the troubleshooting section of the Readme.

Gariantroll, 2016-12-13:
Just an update:  The other programs also have white backgrounds.  I am going to continue with the trouble shooting section and give you an update on my findings.

Gariantroll, 2016-12-13:
Okay, I finally got it to work.  It seems that when I was not overwriting a file properly, so I used sudo and success came soon after.  Thanks a bunch for your help.  I will for sure buy one of your books.  Thanks again!!!

closed 2016-12-14 by hzeller